smuehlst/circle-stdlib #1: CConsole, libusb, libinput - do i really need it?

etet100, 2017-11-18:
I have just upgraded library to v3.0. I wonder if i really need 

CGlueStdioInit(m_FileSystem, **m_Console**);

It tells me to add at least libinput and libusb. I managed to change
it by removing m_Console from libgloss io.cpp. Everything seems to
be working just fine. Do i really need libusb and libinput?

smuehlst, 2017-11-18:
> I have just upgraded library to v3.0...

Please note I have not yet really formally released v3.0, but I have already pushed some changes in preparation of v3.0. I will change the procedure in the future.

> I wonder if i really need
>
> CGlueStdioInit(m_FileSystem, m_Console);

The CConsole reference as well as libusb and libinput are needed for implementing stdin, stdout and stderr for C stdio resp. cin, cout and cerr for C++ iostream.

Apparently you do not really need it, if your code is working fine without it. Probably you are not using any of the mentioned features.

smuehlst, 2017-11-24:
To summarize: If `CGlueStdioInit(m_FileSystem, m_Console)` is not called then the stdio features of the C standard library and the iostream features of the C++ standard library will not work.

etet100, 2017-11-25:
Ok. But i need filesystem working. I want to call GlueStdioInit(&m_FileSystem, nullptr);

jsanchezv, 2017-11-25:
If you don't need the stdio library, you can link with libnosys.a, compile the addon/fatfs library and link with it. Then, you should to use the FATFS library API. That's I've done on my project and works OK.

smuehlst, 2017-11-25:
@etet100: I see, I think a reasonable solution will be to provide two additional overloads of GlueStdioInit() to initialize either the file system or the console part of the stdio implementation.

smuehlst, 2017-11-27:
@etet100: I added the proposed overloads of GlueStdioInit() on the develop branch. You can now also initalize newlib stdio only for using stdin/stdout/stderr, or only for using  the filesystem, or for both as it was already possible. Please let me know whether you are happy with this solution.

etet100, 2017-11-30:
I'm happy and my raspberry is happy. Thank you.

closed 2017-11-30 by etet100