smuehlst/circle-stdlib #2: random crashes (libm?)

etet100, 2017-11-22:
Hi. I'm experiencing random crashes of my app (after 2 seconds or after 2 days). It uses such external libraries as freetype and agg (anti-grain geometry, which extensively uses math). After a few days of guessing i noticed unpredictable behaviour of some math function. Let's take sample project 01-nosys:

```
	m_Logger.Write (FromKernel, LogNotice, "Call acos()");
	double const g = acos (22);
	m_Logger.Write (FromKernel, LogNotice, "acos(22) %f", f);

	//my code
	CString Message;
	while (true) 
	{
			int x = 320 * 240;
			while (x--) {
					uint16_t dest = 1;//(uint16_t)((float)rand() * 2.77f * ((float)rand() * 0.82f) * sin(rand()));
					//float a = sin(dest);
					float b = tan(dest);
					float c = acos(b);// * atan(a);
					float d = sqrt(c);
				Message.Format ("test %.2f\r\n", d);
					//float d = cos(c);
			}
			m_Screen.Write ((const char *) Message, Message.GetLength ());
	}
	//my code

	// Search element in sorted array
	int const nSortedArray[] = { -10, -1, 0, 1, 2, 3, 4, 5, 6 };

```

This code is crashing my raspberry zero 1.3 almost immediately. Do you know why? It looks suspicious to me. When i'm playing with math function (for example changing order) it's sometimes crashing immediately, sometimes after random time, sometimes it's not crashing (but i'm not patient). 

libcircle and libcirclenewlib compiled with gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1)
sample code compiled with gcc 4.9.3 or 6.3.1

smuehlst, 2017-11-22:
I can reproduce that the test program crashes my Raspberry Pi Zero. The screen is filled with a garbage pattern. Is this the same symptom that you have.
The weird thing is that the program runs without problem in QEMU.

etet100, 2017-11-23:
Yes, exactly the same.

rsta2, 2017-11-24:
That issue is caused by Circle itself. Circle does not have a VFP support code yet. In certain cases if floating point instructions are executed with invalid operands (e.g. out of definition range) a floating point exception is generated on the RPi 1, which is basically an "Undefined Instruction" abort.

Because the Circle exception handler was not installed in the sample 01-nosys, the system simply crashed when acos() was called with an invalid operand in your code above. Because of a change in architecture, this exception is not generated on the RPi 2 and 3.

I will provide an update soon, to fix this for the moment. But this cannot be a full VFP support code this time, which is a difficult thing of its own.

smuehlst, 2017-11-24:
@etet100: I synced the develop branch in commit 558186a4b456980cafe18bf1cbf04fa57fb9efc1 with rsta2/circle@2b2191b4baff07eee5313b8bb7a77088bf6aa56f. Now the acos() calls with invalid arguments no longer cause fatal exceptions. Thanks to @rsta2 for the quick analysis and fix.

closed 2017-11-24 by smuehlst