smuehlst/circle-stdlib #4: samples make failing at linking stage

baldwin1974, 2018-03-20:
I am having trouble building the samples.  I've used arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) on a mac, and 6-2017-q1-update on a linux machine (galliumos distro) and they both fail at the linking stage, though with different errors.  Independently I have compiled Circle and built the samples there and have gotten no errors.  I am not sure what I am doing wrong.

On my mac (if it helps, I am submitting this issue only for the mac) I get:

> arm-none-eabi-g++ -march=armv6k -mtune=arm1176jzf-s -marm -mfpu=vfp -mfloat-abi=hard -Wall -fsigned-char -ffreestanding -D__circle__ -DRASPPI=1 -DSTDLIB_SUPPORT=3 -I ../../libs/circle/include -I ../../libs/circle/addon -I ../../libs/circle/app/lib -O2 -g  -I "../../install/arm-none-circle/include" -I "/Users/baldwinrobertson/opt/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/include" -I ../../include -std=c++14 -c -o main.o main.cpp
> arm-none-eabi-g++ -march=armv6k -mtune=arm1176jzf-s -marm -mfpu=vfp -mfloat-abi=hard -Wall -fsigned-char -ffreestanding -D__circle__ -DRASPPI=1 -DSTDLIB_SUPPORT=3 -I ../../libs/circle/include -I ../../libs/circle/addon -I ../../libs/circle/app/lib -O2 -g  -I "../../install/arm-none-circle/include" -I "/Users/baldwinrobertson/opt/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/include" -I ../../include -std=c++14 -c -o kernel.o kernel.cpp
> arm-none-eabi-ld -o kernel.elf -Map kernel.map -T ../../libs/circle/circle.ld ../../libs/circle/lib/startup.o main.o kernel.o    "../../install/arm-none-circle/lib/libm.a" "../../install/arm-none-circle/lib/libc.a" "../../install/arm-none-circle/lib/libcirclenewlib.a" ../../libs/circle/addon/SDCard/libsdcard.a ../../libs/circle/lib/usb/libusb.a ../../libs/circle/lib/input/libinput.a ../../libs/circle/lib/fs/fat/libfatfs.a ../../libs/circle/lib/fs/libfs.a ../../libs/circle/lib/net/libnet.a ../../libs/circle/lib/sched/libsched.a ../../libs/circle/lib/libcircle.a

... after which follows a lengthy series of errors listing undefined references, e.g.: 

> main.o: In function `main':
> /Users/baldwinrobertson/Documents/Code/pi/analyzer/circle-stdlib/samples/02-stdio-hello/main.cpp:24: undefined reference to `__cxa_end_cleanup'
> main.o:(.ARM.extab.text.startup+0x0): undefined reference to `__gxx_personality_v0'
> kernel.o: In function `CKernel::CKernel()':
> /Users/baldwinrobertson/Documents/Code/pi/analyzer/circle-stdlib/samples/02-stdio-hello/kernel.cpp:39: undefined reference to `__cxa_end_cleanup'
> kernel.o:(.ARM.extab+0x0): undefined reference to `__gxx_personality_v0'
> kernel.o:(.ARM.exidx+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
> 
Following all these errors, I get:

> make: *** [kernel.img] Error 1

I tried running the same makes on my linux box, and again I have no trouble compiling and making the samples in circle, and build.bash successfully compiles circle-stdlib ... but again, no luck in making the samples after that.  (In that case, the errors are e.g., arm-none-eabi-ld: cannot find ../../install/arm-none-circle/lib/libm.a: no such file or directory ...)

Does anything jump out at you as to what I am doing wrong?

baldwin1974, 2018-03-20:
Additional info: I uninstalled and reinstalled arm-none-eabi and now I am having a similar problem with linking the samples in circle. If you think that this is really a user error (well, obviously it is) w/r/t arm-none-eabi-ld then go ahead and close this, although if you have any ideas it would be appreciated!

smuehlst, 2018-03-20:
These error messages look somewhat familiar to me, and I believe I have seen them in the past during development. The bad news is that I cannot remember what the cause was at the moment...

baldwin1974, 2018-03-20:
I know the feeling.... 

I nm'd various .a files and found "assertion_failed" after __aeabi_uidiv and __aeabi_uidivmod throughout.  So it looks to me as if the libraries are not being properly compiled after all, but I currently stumped as to why.

rsta2, 2018-03-21:
The C++ standard lib support requires some support from the libraries coming with the toolchain. It does not work with all toolchains. I think the Mac has not been tested with it, because we do not had one.

You can try the options `--softfp` or/and `--no-cpp` of *build.bash*. Some toolchains do not come with hard-float support for the RPi 1 or are not compatible with the C++ support at all. In this case circle-stdlib can be build without C++ support, but you can build sample 01-nosys only with it.

But if you cannot build Circle itself any more, the problem seems to be more severe now. One problem can be a too old version of *make*. You need GNU make 4.0 or later to successfully build Circle. Make sure to enter `./makeall clean` before building Circle, if you have changed something!

baldwin1974, 2018-03-21:
Updating make on my mac (not an easy thing ... I had to temporarily disable the system integrity protection to do so) resulted in successful builds, once I deleted the clones of circle and circle-stdlib.  I just tested one of them on a RPZ and it works. Thanks to both of you for your help!

closed 2018-03-24 by smuehlst