smuehlst/circle-stdlib #5: Error when running make

GiovanniCmpaner, 2018-08-01:
Hi everyone!

When I run make I get the following errors  (just the lasts lines):

`...
In file included from /mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:17:0:
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/include/circle_glue.h:29:16: error: 'constexpr' does not name a type
         static constexpr unsigned GeneralFailure = ~0U;
                ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/include/circle_glue.h:29:16: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/include/circle_glue.h: In member function 'virtual unsigned int CGlueConsole::Read(void*, unsigned int)':
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/include/circle_glue.h:112:95: error: 'nullptr' was not declared in this scope
                         CScheduler * const scheduler = schedulerActive ? CScheduler::Get () : nullptr;
                                                                                               ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/include/circle_glue.h:122:24: error: 'GeneralFailure' was not declared in this scope
   return nResult < 0 ? GeneralFailure : static_cast<unsigned>(nResult);
                        ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/include/circle_glue.h: In member function 'virtual unsigned int CGlueConsole::Write(const void*, unsigned int)':
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/include/circle_glue.h:129:31: error: 'GeneralFailure' was not declared in this scope
          return nResult < 0 ? GeneralFailure : static_cast<unsigned>(nResult);
                               ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp: In constructor '{anonymous}::CircleFile::CircleFile()':
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:39:30: error: 'nullptr' was not declared in this scope
      CircleFile() : mCGlueIO(nullptr) {}
                              ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp: At global scope:
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:43:5: error: 'constexpr' does not name a type
     constexpr unsigned int MAX_OPEN_FILES = 20;
     ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:43:5: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:44:5: error: 'constexpr' does not name a type
     constexpr unsigned int MAX_OPEN_DIRS = 20;
     ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:44:5: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:46:37: error: 'nullptr' was not declared in this scope
     CFATFileSystem *circle_fat_fs = nullptr;
                                     ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:48:24: error: 'MAX_OPEN_FILES' was not declared in this scope
     CircleFile fileTab[MAX_OPEN_FILES];
                        ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:49:24: error: 'MAX_OPEN_DIRS' was not declared in this scope
     _CIRCLE_DIR dirTab[MAX_OPEN_DIRS];
                        ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp: In function 'int {anonymous}::FindFreeFileSlot()':
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:55:23: error: ISO C++ forbids declaration of 'slot' with no type [-fpermissive]
      for (auto const& slot: fileTab)
                       ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:55:29: error: range-based 'for' loops are not allowed in C++98 mode
      for (auto const& slot: fileTab)
                             ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:55:29: error: 'fileTab' was not declared in this scope
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:57:16: error: request for member 'mCGlueIO' in 'slot', which is of non-class type 'const int'
       if (slot.mCGlueIO == nullptr)
                ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:57:28: error: 'nullptr' was not declared in this scope
       if (slot.mCGlueIO == nullptr)
                            ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp: In function 'int {anonymous}::FindFreeDirSlot()':
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:71:23: error: ISO C++ forbids declaration of 'slot' with no type [-fpermissive]
      for (auto const& slot: dirTab)
                       ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:71:29: error: range-based 'for' loops are not allowed in C++98 mode
      for (auto const& slot: dirTab)
                             ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:71:29: error: 'dirTab' was not declared in this scope
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:73:17: error: request for member 'mOpen' in 'slot', which is of non-class type 'const int'
       if (!slot.mOpen)
                 ^
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp: In function 'void {anonymous}::CGlueInitConsole(CConsole&)':
/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/libs/circle-newlib/libgloss/circle/io.cpp:95:33: error: 'fileTab' was not declared in this scope
[... 83 lines cut ...]
make[1]: *** [all] Error 2
make[1]: Leaving directory '/mnt/c/Users/Projetista01/bare_metal/circle-stdlib/build/circle-newlib'
Makefile:14: recipe for target 'newlib' failed
make: *** [newlib] Error 2`

Make version:
`Projetista01@DESENVOLVIMENTO02:/mnt/c/Users/Projetista01/bare_metal/circle-stdlib$ make --version
GNU Make 4.1
Compilado para x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
Licença GPLv3+: GNU GPL versão 3 ou posterior <http://gnu.org/licenses/gpl.html>
Isto é um aplicativo livre: você pode alterá-lo e redistribui-lo livremente.
NÃO HÁ GARANTIAS, exceto o que for permitido por lei.`

Toolchain version:
`Projetista01@DESENVOLVIMENTO02:/mnt/c/Users/Projetista01/bare_metal/circle-stdlib$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

What can I do to solve this?

Thanks,
Giovanni

GiovanniCmpaner, 2018-08-01:
As workaround, I added `-std=c++14` to the following line, before running `./configure`:

https://github.com/smuehlst/circle-stdlib/blob/0d349f6a4ffd08f3beae65d8385f461b3cb67029/configure#L153

Result:
<pre>    CPPFLAGS_FOR_TARGET="-I\"$CIRCLEHOME/include\" -I\"$TOPDIR/include\" -std=c++14" \<pre>

And then `make` ran without errors.

smuehlst, 2018-08-01:
The oldest gcc/g++ version I tested with is 6.3.1, and that compiles C++14 by default. According to its documentation gcc/g++ 5.4.0 compiles C++98 by default, so yes, you will have to add this flag currently for compiling with this older gcc version.
It is probably a good idea to add this flag explicitly to avoid such confusion in the future.

closed 2018-08-02 by GiovanniCmpaner