smuehlst/circle-stdlib #6: Multicore support

Siddhartha123, 2018-10-25:
The multicore sample provided in circle is not compiling successfully with circle-stdlib.
[image] is a screenshot of the error. The name of the sample is 17-fractal.

rsta2, 2018-10-25:
Native Circle samples have to be ported to be build-able with *circle-stdlib*. You find the Circle *sample/17-fractal* attached ported this way. The files *mandelbrot.h* and *mandelbrot.cpp* are not modified from the original sample. The files in the archive must be extracted into the subdirectory *samples/17-fractal* in the *circle-stdlib* tree.

[17-fractal.zip](https://github.com/smuehlst/circle-stdlib/files/2514992/17-fractal.zip)

Siddhartha123, 2018-10-25:
The code you have attached compiles successfully for single core. However, the compilation fails when ARM_ALLOW_MULTI_CORE is defined in circle/include/circle/sysconfig.h.

rsta2, 2018-10-25:
The provided code does build and run here without problems. Have you cleaned everything before build with `./build.bash -c`?

Siddhartha123, 2018-10-26:
Sorry. I was not doing that.  So, I cleaned everything using `./build.bash -c` , then `./configure` and then `make`. Now the code builds successfully . I did not realise that _sysconfig.h_ was part of the Circle library and needed re-compilation after the ARM_ALLOW_MULTI_CORE flag was added. I now understand that every time the source code is changed, the library has to be cleaned compiled again. Thanks a lot for the help!

closed 2018-10-26 by Siddhartha123