smuehlst/circle-stdlib #8: newlib 3.0.0

jsanchezv, 2018-12-03:
Hi Stephan,

I've found that isn't possible to compile C++ applications using the 8.2.1 gcc version (from https://developer.arm.com/open-source/gnu-toolchain/gnu-a/downloads). I guess that the problem can be solved upgrading newlib to 3.0.0. Any plans for this?

smuehlst, 2018-12-06:
Hi José,

can you please be more concrete what the problems are with gcc 8.2.1? What are the reasons that you need to move to gcc 8.2.1?

An upgrade to newlib 3.0.0 is of course possible, but it will probably have to wait until early 2019 when I have more time.

jsanchezv, 2018-12-07:
Hi Stephan,

I'm doing a bit of experimentation with last gcc released by ARM, no more.

When I try to compile an application that uses libstdc++ I get many errors, some related to 'errno' variable by example.

My application compiles perfectly using gcc 7.3.1, so I don't have any problem with waiting until next year. But your circle-stdlib project is a precious gift for Circle and I hope that you want support it for long time.

Thanks by your hard work in circle-stdlib.

jsanchezv, 2018-12-09:
Hi Stephan,

I've found that I can't compile my app using gcc-arm-none-eabi-7-2018-q2-update-linux.tar.bz2 toolchain, I suspect that because is newlib 3.0.0 based. I'm using Linaro toolchain (https://releases.linaro.org/components/toolchain/binaries/7.3-2018.05/arm-eabi/gcc-linaro-7.3.1-2018.05-x86_64_arm-eabi.tar.xz) that is newlib-2.5.0 based.

I guess that ARM toolchain is unusable to compile full libc/libm/libstdc++ apps.

rsta2, 2018-12-09:
I am answering, because I prepared an tested the last release. The samples build and run successfully here, when built with both gcc-arm-none-eabi-7-2018-q2-update-linux.tar.bz2 from the [ARM developer website](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) and with the Linaro toolchain, you have referenced in your comment.

Your application must have special requirements. We need more information. What error messages do you get? Can you provide a test program, which shows the problem?

Yes, the circle-stdlib support depends on the used toolchain, but it is tested and should work at least with the recommended toolchain from the ARM developer website.

jsanchezv, 2018-12-09:
It's easy start to get compilation errors:

1.- Go to samples/02-stdio-hello directory.
2.- Edit kernel.h and add a line with "#include <cstdlib>" and save the file
3.- Compile

rsta2, 2018-12-09:
I tried it with gcc-arm-none-eabi-7-2018-q2-update-linux.tar.bz2. No errors. Are you sure, your build environment is all right? Please be sure to use `./build.bash -c` before `configure` and `make`.

rsta2, 2019-03-01:
There has been a modification in Circle 39, which is included in the latest release of this circle-stdlib project, which should fix include issues. It depended on the kind and order of included headers, if compilation errors arose.

smuehlst, 2019-03-02:
I verified that I can build circle-stdlib v8.0 with gcc-arm-8.2-2019.01-x86_64-arm-eabi downloaded from https://developer.arm.com/open-source/gnu-toolchain/gnu-a/downloads. The four sample programs `01-nosys`, `02-stdio-hello`, `03-stdio-fatfs` and `04-std` run successfully in QEMU.
So apparently it is unnecessary to update newlib currently to a newer version for building with the most recent ARM gcc, and as an update of newlib could potentially have an impact on whether a build with older toolchains is still possible I'm closing this issue without action.

closed 2019-03-02 by smuehlst