smuehlst/circle-stdlib #9: Porting another library

Siddhartha123, 2019-03-24:
I would like to use another library written in c++ with circle-stdlib. I have the source files of the said library. What changes should I make in makefile to build the library in combination with circle ?

smuehlst, 2019-03-24:
This is a good question. There are a lot of compiler options to set that need to match the options that were in use when circle-stdlib was built. One way to find out would be to do a build of circle-stdlib, and then look at what compiler options are set for the sample applications:
```
$ cd samples/04-std
$ make -n
echo "  CPP   main.o"
aarch64-elf-g++ -Wno-parentheses -DAARCH=64 -march=armv8-a -mtune=cortex-a53 -mlittle-endian -mcmodel=small -Wall -fsigned-char -ffreestanding -D__circle__ -DRASPPI=3 -DSTDLIB_SUPPORT=3 -D__VCCOREVER__=0x04000000  -I ../../libs/circle/include -I ../../libs/circle/addon -I ../../libs/circle/app/lib -I ../../libs/circle/addon/vc4 -I ../../libs/circle/addon/vc4/interface/khronos/include -O2 -g -I "../../install/aarch64-none-circle/include" -I "/home/stm/local/gcc-arm-8.2-2019.01-x86_64-aarch64-elf/bin/../lib/gcc/aarch64-elf/8.2.1/include" -I ../../include -std=c++14 -c -o main.o main.cpp
...
```
From this output you can see all the compiler options and the include paths that you need to set for compiling your own library. Probably you need to translate the relative include paths from the `-I` options to absolute pathnames for using them in your own Makefiles.
Another option is that you include the `<circle-stdlib>/Config.mk` and  `<circle-stdlib>/libs/circle/Rules.mk` files in your Makefile and use the relevant Makefile macros like CC, CFLAGS, CPPFLAGS etc. in your Makefile.

Siddhartha123, 2019-03-27:
Thank you for the answer. I followed your first approach but couldn't make the library work. I am trying to integrate [LittlevGL](https://github.com/littlevgl/lvgl) with circle-stdlib. I am unable to figure out the changes I need to make in the makefile of circle-stdlib. Can you suggest the changes to be made in `<circle-stdlib>/Makefile` .

smuehlst, 2019-03-27:
It shouldn't be necessary to make changes to the Makefile of circle-stdlib. You need to change/configure the Makefile of LittlevGL so that it uses the correct compiler options and the include directories of circle, circle-stdlib and circle-newlib.

rsta2, 2019-03-28:
This is very interesting. I'm considering to support LittlevGL in Circle addon and did a quick test. Here is a working Makefile for it for the moment:

```
CIRCLE_STDLIB_DIR = ../circle-stdlib

-include $(CIRCLE_STDLIB_DIR)/Config.mk

NEWLIBDIR  ?= $(CIRCLE_STDLIB_DIR)/install/$(NEWLIB_ARCH)
CIRCLEHOME ?= $(CIRCLE_STDLIB_DIR)/libs/circle
LVGL_DIR   ?= .

include $(LVGL_DIR)/lvgl/lvgl.mk

OBJS	= $(CSRCS:.c=.o)

liblvgl.a: $(OBJS)
	@echo "  AR    $@"
	@rm -f $@
	@$(AR) cr $@ $(OBJS)

include $(CIRCLEHOME)/Rules.mk

INCLUDE += -I $(CIRCLE_STDLIB_DIR)/include -I $(NEWLIBDIR)/include
```

You have to create a new directory and put this file named *Makefile* into it. The path to the circle-stdlib project has to be updated in the first line. From the directory, where the *Makefile* is, then enter the following:

```
git clone https://github.com/littlevgl/lvgl.git
cd lvgl
git checkout v5.3
cd ..
cp lvgl/lv_conf_templ.h lv_conf.h
```

Edit the new file *lv_conf.h* and enable the content by writing a "1" behind the "#if". Now you can configure and build circle-stdlib first from the circle-stdlib directory and then LittlevGL by entering "make" in the directory where the *Makefile* has been saved.

LittlevGL can be built then with circle-stdlib, but porting needs more work, because a LittlevGL screen driver has to be written. For a quick test I used the CScreenDevice::SetPixel() routines, which is not recommended for a real version, because it's slow. This shows the LittlevGL tutorial "objects" running on Circle in AArch64 QEMU:

[image]

Siddhartha123, 2019-03-31:
Thanks a lot @rsta2 . This is exactly what I needed. Hope to see LittlevGL support for circle soon...

closed 2019-03-31 by Siddhartha123

rsta2, 2019-03-31:
@Siddhartha123 You are welcome. LittlevGL support in Circle is [here](https://github.com/rsta2/circle/tree/develop/addon/littlevgl) (on the *develop* branch at the moment). Please see the sample/README for more info.