smuehlst/circle-stdlib #10: circle-stdlib with qemu

Siddhartha123, 2020-01-22:
The CStdlibAppScreen class does not seem to work with LittlevGL. I could get CStdlibAppScreen class to work using the sample 02-stdio-hello. I have found out that the function CLittlevGL::Update() causes DMA error on qemu. 
[image]

What can I do to make LittlevGL work with circle-stdlib on QEMU ?

rsta2, 2020-01-23:
Can you please ensure, that the [LittlevGL display size setting](https://github.com/rsta2/circle/blob/fe24b6bebd1532f2a0ee981af12eaf50cc9e97fb/addon/littlevgl/lv_conf.h#L39) (800x480 by default) matches the size of your emulated frame buffer in QEMU, which is probably 640x480. You can modify the setting in *addon/littlevgl/lv_conf.h* to do this. Otherwise there is an overflow in the calculation of the DMA block stride, which leads to this failed assertion.

I'm afraid, even after correcting this, you will get problems with debugging LittlevGL applications with Circle on QEMU. I tried it with two QEMU versions and both show a distorted display image of the LittlevGL demo program, while the image is all right on a real Raspberry Pi 3B. I can only guess, there is a problem with 2D transfers on the emulated DMA controller in QEMU. Unfortunately not all features, which run on a real RPi can be debugged using QEMU.

rsta2, 2020-01-23:
I've found a bug in QEMU, which caused that distorted display image with LittlevGL on Circle. After applying the following patch to [my QEMU fork](https://github.com/rsta2/qemu) it is working well:

```
diff --git a/hw/dma/bcm2835_dma.c b/hw/dma/bcm2835_dma.c
index c7ce4e4881..da61945b96 100644
--- a/hw/dma/bcm2835_dma.c
+++ b/hw/dma/bcm2835_dma.c
@@ -49,7 +49,7 @@
 static void bcm2835_dma_update(BCM2835DMAState *s, unsigned c)
 {
     BCM2835DMAChan *ch = &s->chan[c];
-    uint32_t data, xlen, ylen;
+    uint32_t data, xlen, xlen_td, ylen;
     int16_t dst_stride, src_stride;
 
     if (!(s->enable & (1 << c))) {
@@ -67,13 +67,13 @@ static void bcm2835_dma_update(BCM2835DMAState *s, unsigned c)
 
         if (ch->ti & BCM2708_DMA_TDMODE) {
             /* 2D transfer mode */
-            ylen = (ch->txfr_len >> 16) & 0x3fff;
-            xlen = ch->txfr_len & 0xffff;
+            ylen = ((ch->txfr_len >> 16) & 0x3fff) + 1;
+            xlen_td = xlen = ch->txfr_len & 0xffff;
             dst_stride = ch->stride >> 16;
             src_stride = ch->stride & 0xffff;
         } else {
             ylen = 1;
-            xlen = ch->txfr_len;
+            xlen_td = xlen = ch->txfr_len;
             dst_stride = 0;
             src_stride = 0;
         }
@@ -112,6 +112,7 @@ static void bcm2835_dma_update(BCM2835DMAState *s, unsigned c)
             if (--ylen != 0) {
                 ch->source_ad += src_stride;
                 ch->dest_ad += dst_stride;
+		xlen = xlen_td;
             }
         }
         ch->cs |= BCM2708_DMA_END;
```

[image]

Siddhartha123, 2020-01-24:
@rsta2 Thanks a lot. I changed the lvgl size setting. Also your patch works perfectly. You should make a pull request to official qemu repo. Appreciate your help a lot !!!

closed 2020-01-24 by Siddhartha123

rsta2, 2020-01-24:
You are welcome. I have submitted the QEMU patch. We will see, if it will be accepted.

rsta2, 2020-02-11:
The QEMU patch has been accepted. It was split into two commits.