smuehlst/circle-stdlib #11: Memory leaks found by static analysis tool

smuehlst, 2020-08-06:
Leaking memory. CAdaptedSocket [:28] is allocated on the heap and never freed. In function mbedtls_net_connect [:21] .

Leaking memory. CAdaptedSocket [:53] is allocated on the heap and never freed. In function mbedtls_net_bind [:41] .

rsta2, 2020-08-07:
I may look like, that memory is leaked here, because the pointer to `pSocket` is saved in a static array in `CAdaptedSocket` and retrieved from there by a handle, which is probably not obvious. It is deleted here:

https://github.com/smuehlst/circle-stdlib/blob/0d349f6a4ffd08f3beae65d8385f461b3cb67029/src/circle-mbedtls/net_sockets.cpp#L169

So I suppose, there is no memory leak here.

smuehlst, 2020-08-08:
I see, a very sophisticated example how to outsmart static code analysis :+1:

closed 2020-08-08 by smuehlst