smuehlst/circle-stdlib #12: can not build samples/01-nosys without c++

hinxx, 2020-10-18:
I understand, as per https://github.com/smuehlst/circle-stdlib/issues/4#issuecomment-374891871 comment, that it should be possible to build first sample even if configuring with `--no-cpp`.
I can build all samples if I use C++ standard library support, so my toolchain should be OK, I guess.

With the following:
```
hinxx@obzen ~/Code/circle-stdlib $ ./configure -n -p /home/hinxx/Projects/baremetal-rpi/gcc-arm-9.2-2019.12-x86_64-arm-none-eabi/bin/arm-none-eabi-
hinxx@obzen ~/Code/circle-stdlib $ cat libs/circle/Config.mk 
RASPPI = 1
PREFIX = /home/hinxx/Projects/baremetal-rpi/gcc-arm-9.2-2019.12-x86_64-arm-none-eabi/bin/arm-none-eabi-
FLOAT_ABI = hard
STDLIB_SUPPORT = 2
STDDEF_INCPATH = "/home/hinxx/Projects/baremetal-rpi/gcc-arm-9.2-2019.12-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.2.1/include"
CFLAGS = -Wno-parentheses 
hinxx@obzen ~/Code/circle-stdlib $ make -j6
```

I get the following error:
```
hinxx@obzen ~/Code/circle-stdlib $ make build-samples 
make -C samples/01-nosys
make[1]: Entering directory '/home/hinxx/Code/circle-stdlib/samples/01-nosys'
make[1]: *** No rule to make target 'cstring', needed by 'main.o'.  Stop.
make[1]: Leaving directory '/home/hinxx/Code/circle-stdlib/samples/01-nosys'
Makefile:31: recipe for target 'build-stdlib-samples' failed
make: *** [build-stdlib-samples] Error 2
```

hinxx, 2020-10-19:
I guess it is [this line](https://github.com/smuehlst/circle-stdlib/blob/1bce5eadab34c13ca33738fc46265868284b5a5b/include/circle_stdlib_app.h#L33) that wants `cstring` causes this.
Is `cstring` needed for the `strcmp()` [here](https://github.com/smuehlst/circle-stdlib/blob/1bce5eadab34c13ca33738fc46265868284b5a5b/include/circle_stdlib_app.h#L183) , only ?

Maybe using `string.h` instead makes it more C friendly?

smuehlst, 2020-10-19:
@hinxx You are right that replacing `<cstring>` with `<string.h>` fixes the build error when configuring with `--no-cpp`. I made that change and pushed it to master. 

Thanks for the suggestion.

closed 2020-10-19 by smuehlst