smuehlst/circle-stdlib #13: building only string handling C library functions

hinxx, 2020-10-18:
I have an library I can compile with circle, but only using circle-stdlib. The library uses some C standard library calls, like sscanf() and atof(), and that is pretty much it. For that purpose, AFAICT, I need to have some sort of C standard library.
As this works with the use of circle-stdlib for me, it does add maintenance effort and I think blows ups the ELF file (TBH, can't tell if it is the library I want to use or addition of the circle-stdlib to the build process). it is not like 16GB SD card can not handle, but still. Last but not least, my compile/serial flash/test runs became quite long as well.

Is there a way to have as really fine tuned C standard library that would hold only cherry picked functions?

I know this might be out of circle/circle-stdlib scope; I'm merely fishing for some pointers and guidance of some sort on how to get this done. Thank you in advance!

rsta2, 2020-10-18:
Not sure, if this is really needed. Normally an application includes only the parts of a library, which are really needed, into its binary image. So the overhead should be small. Maybe it is greater at the moment because of issue #12. It's the task of the linker to decide, which functions have to be included into the binary image.

hinxx, 2020-10-19:
I think you are right.
After giving it a bit more though I would soon need some sort of filesystem access and alike as well, which is what is AFAICT supported in circle-stdlib/newlib.

smuehlst, 2020-10-19:
I agree with @rsta2 that it is up to the linker to include only the modules from the libraries that are actually needed.

If you want to find out what ends up in the executable, you can take a look at the .lst and .map files that are produced as a side effect of linking the executable.

closed 2020-10-19 by smuehlst