smuehlst/circle-stdlib #16: circle samples will not link - many undefined references

sky-flix, 2020-12-18:
I can configure and make in the root circle-stdlib folder without issue. And I even have a working app based on circle-stdlib. However, the circle samples will not link with circle-stdlib/libs/circle set as CIRCLEHOME in their Makefile. If I modify the Makefile to use a cleanly installed circle, they link just fine. Using aarch64-none-elf compiler (9.2.1), if that helps. I've tried including gcc-arm-9.2-2019.12-x86_64-aarch64-none-elf/lib/gcc/aarch64-none-elf/9.2.1/include without any progress.

I'm stumped as to the cause.

sky-flix, 2020-12-19:
I've pinpointed the error down to the:

STDLIB_SUPPORT = 3

in the Config.mk. If I don't include standard library support, the samples compile and link fine.

rsta2, 2020-12-20:
The circle-stdlib project provides C and C++ standard library support for Circle-based applications. It is not intended to build native Circle sample programs in the circle-stdlib build environment. If you want to use standard library support in your application and use a native Circle sample program as basis, the CKernel class and the Makefile of the sample have to be modified.

sky-flix, 2020-12-20:
aah, yes, makes perfect sense. I tweaked the Makefile cflags to include the newlib and compiler includes, and everything links just fine now. Thanks.

closed 2020-12-20 by sky-flix