smuehlst/circle-stdlib #20: Can't build project

leiradel, 2021-03-04:
Hi, I needed some stdlib functions for my Circle project and was directed here. My project was building with Circle (before I added the stdlib functions), but now it refuses to build.

```
$ make -j6
/home/leiradel/Develop/circle-stdlib/install/aarch64-none-circle/../../libs/circle/Rules.mk:74: *** RASPPI must be set to 3 or 4.  Stop.
```

I've tried some things to specify `RASPPI=3` but none worked. My Makefile is

```makefile
include ../../Config.mk

NEWLIB_ARCH = aarch64-none-circle
NEWLIBDIR = /home/leiradel/Develop/circle-stdlib/install/$(NEWLIB_ARCH)
CIRCLEHOME = $(NEWLIBDIR)/../../libs/circle

OBJS = \
	frontend/Main.o frontend/Audio.o frontend/Video.o frontend/Fifo.o \
	core/main.o \
	speex/resample.o \
	lrcpp/src/Components.o lrcpp/src/Core.o lrcpp/src/CoreFsm.o lrcpp/src/Frontend.o

include $(CIRCLEHOME)/Rules.mk

CFLAGS += -I$(NEWLIBDIR)/include -I$(STDDEF_INCPATH) -I$(NEWLIBDIR)/../../include
LIBS := $(NEWLIBDIR)/lib/libm.a $(NEWLIBDIR)/lib/libc.a $(NEWLIBDIR)/lib/libcirclenewlib.a \
	$(CIRCLEHOME)/addon/vc4/sound/libvchiqsound.a \
	$(CIRCLEHOME)/addon/vc4/vchiq/libvchiq.a \
	$(CIRCLEHOME)/addon/linux/liblinuxemu.a \
	$(CIRCLEHOME)/lib/sched/libsched.a \
	$(CIRCLEHOME)/lib/libcircle.a
```

This is likely an issue with my project, but I have no other place to ask.

Thanks in advance.

rsta2, 2021-03-05:
Have you called `./configure` from circle-stdlib root? This script creates two *Config.mk* files, one in circle-stdlib root and one in libs/circle. The one in circle-stdlib root will be included by the first statement in your project's *Makefile*. The second will be read from *Rules.mk*, which is included into the Makefile too.

This error message comes from *Rules.mk*, which expects the `RASPPI` variable set to 3 or 4 for AArch64 operation in *libs/circle/Config.mk*. This is normally also written by the `configure` script. It is unusual to define `NEWLIB_ARCH` in the project's *Makefile*. It is written by the `configure` script to *Config.mk* in circle-stdlib root.

leiradel, 2021-03-05:
> Have you called ./configure from circle-stdlib root?

Yes, I've called:

```
$ ./configure --prefix=aarch64-linux-gnu-
$ make RASPPI=3
```

It built successfully, it's just when I try to build my project that it complains about `RASPPI` not being set.

smuehlst, 2021-03-05:
You should specify the desired Rasperry Pi model as an argument to the configure script, e.g.:

    ./configure --prefix=aarch64-linux-gnu- -r 3
    make

Did you try that?

leiradel, 2021-03-05:
> Did you try that?

Thanks, that did the trick, I used `make RASPPI=3` when it didn't work.

Now I'm getting one duplicated symbol and a bunch of missing symbols:

```
$ make
  LD    kernel8.elf
aarch64-linux-gnu-ld: ../../addon/linux/liblinuxemu.a(sprintf.o): in function `vsnprintf':
/home/leiradel/Develop/circle/addon/linux/sprintf.cpp:46: multiple definition of `vsnprintf'; ../../../circle-stdlib/install/aarch64-none-circle/lib/libc.a(lib_a-vsnprintf.o):vsnprintf.c:(.text+0x0): first defined here
aarch64-linux-gnu-ld: snes9x2002/src/memmap.o: in function `LoadROM':
/home/leiradel/Develop/circle/sample/libretro/snes9x2002/src/memmap.c:357: undefined reference to `__ctype_b_loc'
aarch64-linux-gnu-ld: snes9x2002/src/fxinst.o: in function `printf':
/usr/aarch64-linux-gnu/include/bits/stdio2.h:107: undefined reference to `__printf_chk'
aarch64-linux-gnu-ld: /usr/aarch64-linux-gnu/include/bits/stdio2.h:107: undefined reference to `__printf_chk'
aarch64-linux-gnu-ld: /usr/lib/gcc-cross/aarch64-linux-gnu/9/libstdc++.a(vterminate.o): in function `__gnu_cxx::__verbose_terminate_handler()':
(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x4c): undefined reference to `stderr'
aarch64-linux-gnu-ld: (.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x74): undefined reference to `stderr'
aarch64-linux-gnu-ld: (.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0xa4): undefined reference to `stderr'
aarch64-linux-gnu-ld: (.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0xe0): undefined reference to `stderr'
aarch64-linux-gnu-ld: (.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0xf0): undefined reference to `stderr'
aarch64-linux-gnu-ld: /usr/lib/gcc-cross/aarch64-linux-gnu/9/libstdc++.a(vterminate.o):(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x10c): more undefined references to `stderr' follow
aarch64-linux-gnu-ld: /usr/lib/gcc-cross/aarch64-linux-gnu/9/libgcc_eh.a(unwind-dw2-fde-dip.o): in function `_Unwind_Find_FDE':
(.text+0x1930): undefined reference to `dl_iterate_phdr'
make: *** [../../../circle-stdlib/libs/circle/Rules.mk:155: kernel8.img] Error 1
```

Any idea how to fix those?

rsta2, 2021-03-06:
This is odd, because in the normal Circle configuration for circle-stdlib with C++ support the macro STDLIB_SUPPORT has the value 3. With this setting there is no symbol `vsnprintf` in [addon/linux/sprintf.cpp](https://github.com/rsta2/circle/blob/master/addon/linux/sprintf.cpp#L6), but your build log says there are multiple definitions of it with one in *addon/linux/sprintf.cpp*. Please cleanup everything and configure and build again:

```
# from circle-stdlib root:
make mrproper

# these directories must be built and cleaned manually
cd libs/circle/addon/linux
make clean
cd ../vc4
./makeall clean
cd ../../../..

./configure -r 3 -p aarch64-linux-gnu-
make -j

cd libs/circle/addon/linux
make
cd ../vc4
./makeall --nosample -j
cd ../../../..

# now clean and build your own project
```

BTW. We are using the toolchain for the bare-metal target aarch64-none-elf-. Please see the *Prerequisites* section in README.md for a link. I do not know, if the aarch64-linux-gnu toolchain will work in all cases.

leiradel, 2021-03-07:
> BTW. We are using the toolchain for the bare-metal target aarch64-none-elf-

That may be the issue. I'll install that toolchain and try again. Thanks for the tip.

smuehlst, 2021-03-19:
Closing because no further feedback arrived.

closed 2021-03-19 by smuehlst