smuehlst/circle-stdlib #21: Building trouble

l-o-l, 2021-04-05:
Hi,

I'm trying to build on the Pi itself and am having some problems (I'm new to PI and Linux):

What I've tried:

./configure -r 4 -p ''
make

This part seems to work, or at least there are no obvious errors. When I go to build the samples however, I get the following:

make build-stdlib-samples
make -C samples/01-nosys
make[1]: Entering directory '/home/pi/src/circle-stdlib/samples/01-nosys'
  LD    kernel7l.elf
ld: /usr/lib/gcc/arm-linux-gnueabihf/8/libstdc++.a(eh_globals.o): in function `__cxa_get_globals_fast':
(.text.__cxa_get_globals_fast+0xc): undefined reference to `__tls_get_addr'
ld: /usr/lib/gcc/arm-linux-gnueabihf/8/libstdc++.a(eh_globals.o): in function `__cxa_get_globals':
(.text.__cxa_get_globals+0xc): undefined reference to `__tls_get_addr'
ld: /usr/lib/gcc/arm-linux-gnueabihf/8/libstdc++.a(vterminate.o): in function `__gnu_cxx::__verbose_terminate_handler()':
(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x190): undefined reference to `stderr'
make[1]: *** [../../libs/circle/Rules.mk:156: kernel7l.img] Error 1
make[1]: Leaving directory '/home/pi/src/circle-stdlib/samples/01-nosys'
make: *** [Makefile:31: build-stdlib-samples] Error 2

Any help would be greatly appreciated!

rsta2, 2021-04-05:
The C++ standard lib support in circle-stdlib is unfortunately not compatible with the standard toolchain of Raspberry Pi OS. You can try to add the option `--no-cpp` to `configure` to have the C standard library support at least.

smuehlst, 2021-04-05:
The circle-stdlib build is only tested with the cross compiler hosted on Intel Linux. But it looks like that there's a AArch64-hosted toolchain available on the [GNU-A downloads page](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads) as well:

https://developer.arm.com/-/media/Files/downloads/gnu-a/10.2-2020.11/binrel/gcc-arm-10.2-2020.11-aarch64-arm-none-eabi.tar.xz

Maybe you can give this one a try.

l-o-l, 2021-04-18:
Just as an update. I've installed manjaro 64 bit on my Pi, used the suggested tool chain, and everything now builds fine! Thanks for the help on this.

closed 2021-04-18 by l-o-l

smuehlst, 2021-04-18:
> Just as an update. I've installed manjaro 64 bit on my Pi, used the suggested tool chain, and everything now builds fine! Thanks for the help on this.

You're welcome. I will add a note to the README.md file that a build under Linux on the Raspberry Pi is possible with the corresponding toolchain.

l-o-l, 2021-04-18:
A 64bit version of the OS is required (due to the tool chain only existing in a 64bit version). I was able to run the 04-std sample, so everything is good.

Xalior, 2021-05-02:
Can confirm that `gcc-arm-10.2-2020.11-aarch64-arm-none-eabi` works great on the beta of Raspberry Pi OS 64bit.

I extracted into the `circle-stdlib` folder (in a `gcc-arm-10.2-2020.11-aarch64-arm-none-eabi` subfolder) and just ran `export PATH=`pwd`/gcc-arm-10.2-2020.11-aarch64-arm-none-eabi/bin:$PATH` and everything built like a charm.  Thanks for the tip. <3 x