smuehlst/circle-stdlib #23: Compilation fails on FreeBSD

probonopd, 2022-02-05:
Does circle-stdlib have some hidden Linux dependencies?

I tried to compile it on FreeBSD but ran into some issues. I am not saying that FreeBSD support is a must have but maybe my notes can be useful for removing unintended Linux-isms and/or for others trying to compile it on FreeBSD.

My findings:
* FreeBSD by default does not use GNU make (but GNU make can be used by installing and using the `gmake` command instead). I have not found a proper way to tell the circle build system to use `gmake` instead of `make` throughout
* For the toolchain one has to use the `devel/gcc-arm-embedded` FreeBSD package, __not__ `devel/arm-none-eabi-gcc`, and one has to `export PATH=/usr/local/gcc-arm-embedded-10-2020-q4-major/bin/:$PATH`
* Compilation fails with an error I could not resolve: `circle-stdlib/libs/circle/include/circle/types.h:50:16: error: conflicting declaration 'typedef long int ssize_t'`

```
cd ../circle-stdlib
./configure -r 4 # Raspberry Pi 4
# (...)
# configure: creating ./config.status
# config.status: creating Makefile

gmake
# cd libs/circle && ./makeall --nosample
# make[1]: "/usr/home/user/MiniDexed/circle-stdlib/libs/circle/tools/Makefile" line 5: Need an operator
# (...)
# make[1]: Fatal errors encountered -- cannot continue
# make[1]: stopped in /usr/home/user/MiniDexed/circle-stdlib/libs/circle/tools
# gmake: *** [Makefile:10: circle] Error 1
```

Probably this is related to me trying to compile on FreeBSD. So:

```
sed -i -e 's|-make|-gmake|g' ./libs/circle/makeall
```

Compilation proceeds much further but then stops with

```
gmake[1]: Entering directory '/usr/home/user/MiniDexed/circle-stdlib/libs/circle/addon/fatfs'
In file included from /usr/include/c++/v1/stdint.h:106,
                 from ff.h:44,
                 from ffsystem.cpp:8:
/usr/include/c++/v1/__config:1141:6: error: #error "No thread API"
 #    error "No thread API"
      ^~~~~
In file included from /usr/include/c++/v1/stdint.h:106,
                 from ff.h:44,
                 from diskio.cpp:11:
/usr/include/c++/v1/__config:1141:6: error: #error "No thread API"
 #    error "No thread API"
      ^~~~~
  CC    ff.o
  CPP   diskio.o
In file included from /usr/include/c++/v1/stdint.h:106,
                 from ff.h:44,
                 from diskio.cpp:11:
/usr/include/c++/v1/__config:1141:6: error: #error "No thread API"
 #    error "No thread API"
      ^~~~~
gmake[1]: *** [../../Rules.mk:143: diskio.o] Error 1
gmake[1]: Leaving directory '/usr/home/user/MiniDexed/circle-stdlib/libs/circle/addon/fatfs'
gmake: *** [Makefile:12: circle] Error 2
```

https://forums.freebsd.org/threads/raspberry-pi-pico.79394/page-2 advises deleting the `arm-none-eabi` package and replacing it with `gcc-arm-embedded` but when I do this I get:

```
./configure
# Error: Invalid toolchain prefix, exiting
# TOOLPREFIX is "arm-none-eabi-"
```

On we go:

```
export PATH=/usr/local/gcc-arm-embedded-10-2020-q4-major/bin/:$PATH
./configure
gmake
# (...)
gmake[1]: Entering directory '/usr/home/user/MiniDexed/circle-stdlib/libs/circle/addon/qemu'
  CPP   qemuhostfile.o
  AR    libqemusupport.a
gmake[1]: Leaving directory '/usr/home/user/MiniDexed/circle-stdlib/libs/circle/addon/qemu'
cd libs/circle/addon/wlan && ./makeall --nosample
make[1]: "/usr/home/user/MiniDexed/circle-stdlib/libs/circle/Rules.mk" line 49: Need an operator
(...)
make[1]: Fatal errors encountered -- cannot continue
make[1]: stopped in /usr/home/user/MiniDexed/circle-stdlib/libs/circle/addon/wlan
gmake: *** [Makefile:14: circle] Error 1
```

It seems like it is using `make` instead of `gmake` again...

So getting desparate now, this is most likely __NOT the correct way to do this__...

```
sudo mv /usr/bin/make /usr/bin/make.original
sudo ln -s /usr/local/bin/gmake /usr/local/bin/make
```

Compilation continues but eventually ends with.

```
arm-none-eabi-g++ -B/usr/home/user/MiniDexed/circle-stdlib/build/circle-newlib/arm-none-circle/newlib/ -isystem /usr/home/user/MiniDexed/circle-stdlib/build/circle-newlib/arm-none-circle/newlib/targ-include -isystem /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/newlib/libc/include -B/usr/home/user/MiniDexed/circle-stdlib/build/circle-newlib/arm-none-circle/libgloss/arm -L/usr/home/user/MiniDexed/circle-stdlib/build/circle-newlib/arm-none-circle/libgloss/libnosys -L/home/user/MiniDexed/circle-stdlib/libs/circle-newlib/libgloss/arm    -Wno-parentheses -I/home/user/MiniDexed/circle-stdlib/libs/circle/include -I/home/user/MiniDexed/circle-stdlib/libs/circle/addon -I/home/user/MiniDexed/circle-stdlib/include -I. -I/home/user/MiniDexed/circle-stdlib/libs/circle-newlib/libgloss/circle/.. -c -Wno-parentheses -fno-exceptions -fno-rtti /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/libgloss/circle/gettod.cpp
In file included from /home/user/MiniDexed/circle-stdlib/libs/circle/include/circle/exceptionstub.h:24,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle/include/circle/interrupt.h:24,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle/include/circle/timer.h:23,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/libgloss/circle/gettod.cpp:15:
/home/user/MiniDexed/circle-stdlib/libs/circle/include/circle/types.h:49:24: error: conflicting declaration 'typedef long unsigned int size_t'
   49 | typedef unsigned long  size_t;
      |                        ^~~~~~
In file included from /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/newlib/libc/include/sys/cdefs.h:47,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/newlib/libc/include/sys/time.h:43,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/libgloss/circle/gettod.cpp:8:
/usr/local/gcc-arm-embedded-10-2020-q4-major/lib/gcc/arm-none-eabi/10.2.1/include/stddef.h:209:23: note: previous declaration as 'typedef unsigned int size_t'
  209 | typedef __SIZE_TYPE__ size_t;
      |                       ^~~~~~
In file included from /home/user/MiniDexed/circle-stdlib/libs/circle/include/circle/exceptionstub.h:24,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle/include/circle/interrupt.h:24,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle/include/circle/timer.h:23,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/libgloss/circle/gettod.cpp:15:
/home/user/MiniDexed/circle-stdlib/libs/circle/include/circle/types.h:50:16: error: conflicting declaration 'typedef long int ssize_t'
   50 | typedef long   ssize_t;
      |                ^~~~~~~
In file included from /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/newlib/libc/include/sys/time.h:45,
                 from /home/user/MiniDexed/circle-stdlib/libs/circle-newlib/libgloss/circle/gettod.cpp:8:
/home/user/MiniDexed/circle-stdlib/libs/circle-newlib/newlib/libc/include/sys/types.h:184:18: note: previous declaration as 'typedef _ssize_t ssize_t'
  184 | typedef _ssize_t ssize_t;
      |                  ^~~~~~~
make[4]: *** [Makefile:100: gettod.o] Error 1
```

smuehlst, 2022-02-05:
> Does circle-stdlib have some hidden Linux dependencies?

Probably. From my side a build has never been tested on other platforms apart from Linux.

The build errors related to make/gmake happen in the Circle build that is started from the circle-stdlib build, so you would have to ask @rsta2 about that. I guess `make` needs to be replaced with `$(MAKE)` in all the Makefiles to fix this.

I will later today take a look at where the conflicting types come from. The `configure` script has an option `-s <path>, --stddefpath <path>` that might be relevant for determining the location of the correct `stddef.h` header. It may also be a problem caused by a different gcc cross compiler on FreeBSD.

smuehlst, 2022-02-05:
These type redefinition problems sound similar to what was described here:

https://github.com/rsta2/circle/issues/97

Maybe there's somewhere a 32-bit/64-bit mismatch.

probonopd, 2022-02-05:
Thanks, will build on Linux for now. Your decision whether you want to keep this ticket open.

smuehlst, 2022-02-05:
I have personally no ambitions to make the build work under FreeBSD, but of course I would accept corresponding pull requests.

closed 2022-02-05 by smuehlst

probonopd, 2022-02-05:
OK, thanks for the clarification.