smuehlst/circle-stdlib #25: Are threads supported?

tly000, 2022-08-06:
Hi, does this project support threads, i.e. pthreads or std::threads? 

I tried to port a library to circle that uses threads, mutexes and condition variables, but i got a lot of compile errors. If i understand the problem correctly this cannot be solved at the moment because the libgcc that is included in the ARM toolchain does not support threads. I think to solve it libgcc, libstdc++ and circle-newlib must be built from source with the pthread from circles linux emulation. 

Is this correct or am i doing something wrong?

Thanks!

rsta2, 2022-08-07:
There is some very limited pthreads support in *libs/circle/addon/linux/*, which has been implemented while porting the VC4 accelerated graphics userland code. That's currently all.

tly000, 2022-08-07:
Thanks for the response. Ill try to work around the errors for now:

closed 2022-08-07 by tly000