smuehlst/circle-stdlib #26: Usage guidance?

willdieh, 2022-08-23:
Greetings and thank you very much for this excellent project.

I was just wondering if it might be possible to have a paragraph or two in the README for this project on how to best set up the circle-stdlib environment for compilation? I've been approaching it blindly and have had decent success but am not sure where to draw the line on implementation libraries. Right now I include the circle-stdlib libraries/include directoreis, the circle libraries/include directories (including the addon stuff like Fatfs and USB needed for circle-stdlib's CStdlibAppStdio class, followed finally by the include directories of the baremetal development version of gcc provided by ARM. Should we also be using -specs=nosys.spec or does libcirclenewlib provide the needed low level functions for the raspberry pi? 

Thanks again for your amazing work!

willdieh, 2022-08-23:
Here is my current env (set in .bashrc)

`LIBS=-Wl,--start-group /circle-stdlib/install/arm-none-circle/lib/libm.a /circle-stdlib/install/arm-none-circle/lib/libc.a /circle-stdlib/install/arm-none-circle/lib/libcirclenewlib.a /circle-stdlib/libs/circle/addon/SDCard/libsdcard.a /circle-stdlib/libs/circle/lib/usb/libusb.a /circle-stdlib/libs/circle/lib/input/libinput.a /circle-stdlib/libs/circle/addon/fatfs/libfatfs.a /circle-stdlib/libs/circle/lib/fs/libfs.a /circle-stdlib/libs/circle/lib/net/libnet.a /circle-stdlib/libs/circle/lib/sched/libsched.a /circle-stdlib/libs/circle/lib/libcircle.a /gcc/arm-none-eabi/lib/thumb/v8-a+simd/hard/libstdc++.a /gcc/lib/gcc/arm-none-eabi/11.3.1/thumb/v8-a+simd/hard/libgcc.a /circle-stdlib/libs/circle/addon/vc4/interface/bcm_host/libbcm_host.a /circle-stdlib/libs/circle/addon/vc4/interface/khronos/libkhrn_client.a /circle-stdlib/libs/circle/addon/vc4/interface/vmcs_host/libvmcs_host.a /circle-stdlib/libs/circle/addon/vc4/vchiq/libvchiq.a /circle-stdlib/libs/circle/addon/vc4/interface/vcos/libvcos.a /circle-stdlib/libs/circle/addon/vc4/interface/vmcs_host/libvmcs_host.a /circle-stdlib/libs/circle/addon/linux/liblinuxemu.a -Wl,--end-group

CXXFLAGS=-Wall -DAARCH=32 -mcpu=cortex-a53 -marm -mfpu=neon-fp-armv8 -mfloat-abi=hard -fsigned-char -nostdlib -DNO_CALIBRATE_DELAY -DARM_ALLOW_MULTI_CORE -D__circle__=440500 -DRASPPI=3 -DSTDLIB_SUPPORT=3 -D__VCCOREVER__=0x04000000 -U__unix__ -U__linux__ -O2 -g -T/circle-stdlib/libs/circle/circle.ld -std=c++17
LDFLAGS=-L/circle-stdlib/libs/circle/addon/vc4/interface/khronos -L/circle-stdlib/libs/circle/addon/linux -L/circle-stdlib/libs/circle/addon/vc4/interface/bcm_host

CPPFLAGS=-I/circle-stdlib/libs/circle/include -I/circle-stdlib/libs/circle/addon -I/circle-stdlib/libs/circle/app/lib -I/circle-stdlib/libs/circle/addon/vc4 -I/circle-stdlib/libs/circle/addon/vc4/interface/khronos/include -I/circle-stdlib/install/arm-none-circle/include -I/gcc/lib/gcc/arm-none-eabi/11.3.1/include -I/circle-stdlib/include -I/circle-stdlib/libs/circle/addon/linux

CFLAGS=-Wall -DAARCH=32 -mcpu=cortex-a53 -marm -mfpu=neon-fp-armv8 -mfloat-abi=hard -fsigned-char -nostdlib -DNO_CALIBRATE_DELAY -DARM_ALLOW_MULTI_CORE -D__circle__=440500 -DRASPPI=3 -DSTDLIB_SUPPORT=3 -D__VCCOREVER__=0x04000000 -U__unix__ -U__linux__ -O2 -g -T/circle-stdlib/libs/circle/circle.ld
`

willdieh, 2022-08-23:
I should also note that the basis for this question arises from my attempt to get pthreads support in a library I'm trying to port over (I realize the pthread support in ./libs/circle/addon/linux is limited). circle-stdlib and the circle addon linux emulation both seem to provide pthread types and they conflict. I ended up commenting out the circle-stdlib pthread references in order to rely completely on circle's but wasn't sure if this was the correct approach.

smuehlst, 2022-08-23:
I admit that the instructions for building in the README.md file are not very elaborate, but normally they are sufficient:

https://github.com/smuehlst/circle-stdlib#building-the-libraries

It should not be necessary to set any variables in the environment. The samples with their Makefiles demonstrate how to build applications that are compiled and linked against circle-stdlib and Circle itself.

Regarding pthreads support, so far nothing is implemented in circle-stdlib. So you are right that you should rely on the pthread support that is implemented in Circle directly. I will check whether I can disable the installation of the `pthread.h` header from Newlib.

closed 2022-12-03 by smuehlst