smuehlst/circle-stdlib #27: Redirection does not appear to be supported

kevinboone, 2023-01-25:
There doesn't seem to be a way to redirect stdin/stdout/stderr to a file. dup/dup2 fail at link time -- presumably there is not implementation. `freopen ("foo.txt", "w", stdout)` appears to work, and no further output is produced to the console. But, so far as I can tell, nothing is written to the file, either.  

This is a nuisance for me because I have a number of large libraries that write to stdout, and I need them to process their output in my code. On Linux, I just redirect the output, run the library, and then read back the file. I can't see a way to do this using circle-stdlib.

While it would be nice if dup/dup2 worked, for my purposes I would be perfectly happy to use circle-specific APIs to do the redirection.

smuehlst, 2023-01-25:
I'm sorry, this is one of the known gaps, and I currently am not investing much time into circle-stdlib. There are some more newly implemented functions already waiting on the `develop` branch, which I will probably release in the near future, but `dup()` and `fcntl()` are not among them.

smuehlst, 2023-01-29:
@kevinboone It was less effort than I thought. `dup()`, `dup2()` and the duplication-related functionality of `fcntl()` are now implemented on the `develop` branch as of commit 25aab9c357e5f01a2b8ccb2993bc619e34167944. Please give it a try and let me know whether it works for you.

kevinboone, 2023-01-30:
Hi. Thanks for this. I'm away from my desk at present, but I'm keen to try this, and the chdir/getcwd fix, as soon as I can. Now, I hope you'll pardon my lack of knowledge of GitHub -- if I check out the 'develop' branch of circle-stdlib, will I get the 'develop' branch of Circle as well? Or do I need to check that out separately? If you don't have time to answer this, don't worry -- I'll figure it out by trial-and-error ;)

smuehlst, 2023-01-30:
The following should work from the root directory of the `circle-stdlib` repository:

```
git checkout develop
git submodule update --init --recursive
```

Then proceed with the normal `configure` and `make` procedure.

smuehlst, 2023-02-01:
[circle-stdlib v16](https://github.com/smuehlst/circle-stdlib/tree/v16) has been released where `dup()` and `dup2()` have been implemented.

closed 2023-02-01 by smuehlst

kevinboone, 2023-02-06:
Thank you for implementing this. For the record, I can now redirect stdout by doing something like this:

int fd = open ("foo.txt", O_WRONLY | O_TRUNC | O_CREA);
int stdout_copy = dup (1);
dup2 (fd, 1);
// Do something...
fflush (stdout);
dup2 (stdout_copy, 1);
close (stdout_copy);
close (fd);

smuehlst, 2023-02-07:
You're welcome. Also redirection via `freopen()` does work now.