smuehlst/circle-stdlib #30: It would nice if filesystem synchronization behaviour could be controlled

kevinboone, 2023-02-06:
So far as I can tell by looking at the source code, circle-stdlib file handling functions call `CFatFilesystem::Sychronize()` when closing a file. 

For applications where shut-down might be disorderly (battery flat, for example), it would be good if the filesystem could be synchronized more often. In fact, it would be good if it could be done after every operation that writes any data, or renames, or deletes a file. This would be like mounting a filesystem using the 'sync' option in Linux.

Of course, if power is lost whilst writing a long file, it has to be accepted that the file will be incomplete. However, it would be good to be able to ensure that the filesystem itself could not be broken.

I could implement this change in circle-stdlib, but it might be something that is better implemented in circle itself. I logged a new issue for circle:

https://github.com/rsta2/circle/issues/348

kevinboone, 2023-02-08:
I think that 99% of the work could be done if, every time something calls f_write(), it is immediately followed by a call to f_sync(). To be honest, I'm not sure how your implementation links ARM newlib with ChaN's FATFS code. When I've used Newlib in the past, I've just linked the appropriate ARM library, and provided implementations of the stubs _write(), _read(), etc., in my own code. So I would have my _write() implementation call f_write() then f_sync(). However, I don't think your implementation works like this, and I can't really figure out where f_write() gets called. 

An additional problem, I think, would be making it configurable. I guess most people would not want this 'feature'. Perhaps it could be switched on using a flag on the kernel command line?

The real problem, though, would be testing it. I think it would be very difficult to prove whether it actually makes any difference to shut-down behaviour.

If nobody else is interested in this, don't worry about it. It's not that important.

smuehlst, 2023-02-08:
> ... I can't really figure out where f_write() gets called.

`f_write()` is called from the Circle-specific part of [circle-newlib](https://github.com/smuehlst/circle-newlib):

https://github.com/smuehlst/circle-newlib/blob/5fb5e151f6b5c4bd24d471dd0b6d09e43ab4deb4/libgloss/circle/io.cpp#L432-L434

```
auto const fresult = f_write (&mFile, pBuffer,
                              static_cast<UINT> (nCount),
                              &bytesWritten);
```

smuehlst, 2023-02-25:
> If nobody else is interested in this, don't worry about it. It's not that important.

I think we can close this because of lack of further activity.

closed 2023-02-25 by smuehlst