smuehlst/circle-stdlib #32: How to build LVGL without samples?

dolence, 2024-02-09:
I'm trying to add to LVGL to Minidexed but I couldn't finish the buiding proccess:

I have made a copy of _lv_honf.h_ to Minidexed _src_ dir but I'm not sure if it is being used isntead of the one on circle-stdlib LVGL directory.

rsta2, 2024-02-09:
To add LVGL support, you must checkout the submodule in *circle-stdlib/libs/circle/addon/lvgl*, make it, add the CLVGL class to your project and initialize it properly by calling `CLVGL::Initialize()`. Then you must call `CLVGL::Update()` in the main loop of your application. If you want to modify *lv_conf.h*, you must do this in the directory mentioned above.

This is a sample program, which adds LVGL support to a circle-stdlib application:

[lvgltest.zip](https://github.com/smuehlst/circle-stdlib/files/14223260/lvgltest.zip)

dolence, 2024-02-09:
Thank you for taking your time to help me through this. It still throwing errors, tho. Funny thing is that I managed to run LVGL on circle without the MiniDexed files before.

```
ake[1]: Entering directory '/home/dolence/MiniDexed/circle-stdlib/libs/circle/addon/lvgl'
make[1]: 'liblvgl.a' is up to date.
make[1]: Leaving directory '/home/dolence/MiniDexed/circle-stdlib/libs/circle/addon/lvgl'
make -C src -j
make[1]: Entering directory '/home/dolence/MiniDexed/src'
  CPP   main.o
  CPP   kernel.o
kernel.cpp: In constructor 'CKernel::CKernel()':
kernel.cpp:38:29: error: no matching function for call to 'CScreenDevice::CScreenDevice()'
   38 |  m_GUI(&mScreen, &mInterrupt)
      |                             ^
In file included from circle_stdlib_app.h:23,
                 from kernel.h:23,
                 from kernel.cpp:20:
../circle-stdlib/libs/circle/include/circle/screen.h:158:2: note: candidate: 'CScreenDevice::CScreenDevice(unsigned int, unsigned int, boolean, unsigned int)'
  158 |  CScreenDevice (unsigned nWidth, unsigned nHeight, boolean bVirtual = FALSE,
      |  ^~~~~~~~~~~~~
../circle-stdlib/libs/circle/include/circle/screen.h:158:2: note:   candidate expects 4 arguments, 0 provided
../circle-stdlib/libs/circle/include/circle/screen.h:151:7: note: candidate: 'constexpr CScreenDevice::CScreenDevice(const CScreenDevice&)'
  151 | class CScreenDevice : public CDevice /// Writing characters to screen
      |       ^~~~~~~~~~~~~
../circle-stdlib/libs/circle/include/circle/screen.h:151:7: note:   candidate expects 1 argument, 0 provided
kernel.cpp:38:29: error: no matching function for call to 'CLVGL::CLVGL(CWriteBufferDevice*, CInterruptSystem*)'
   38 |  m_GUI(&mScreen, &mInterrupt)
      |                             ^
In file included from kernel.h:29,
                 from kernel.cpp:20:
../circle-stdlib/libs/circle/addon/lvgl/lvgl.h:39:2: note: candidate: 'CLVGL::CLVGL(CBcmFrameBuffer*, CInterruptSystem*)'
   39 |  CLVGL (CBcmFrameBuffer *pFrameBuffer, CInterruptSystem *pInterrupt);
      |  ^~~~~
../circle-stdlib/libs/circle/addon/lvgl/lvgl.h:39:26: note:   no known conversion for argument 1 from 'CWriteBufferDevice*' to 'CBcmFrameBuffer*'
   39 |  CLVGL (CBcmFrameBuffer *pFrameBuffer, CInterruptSystem *pInterrupt);
      |         ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
../circle-stdlib/libs/circle/addon/lvgl/lvgl.h:38:2: note: candidate: 'CLVGL::CLVGL(CScreenDevice*, CInterruptSystem*)'
   38 |  CLVGL (CScreenDevice *pScreen, CInterruptSystem *pInterrupt);
      |  ^~~~~
../circle-stdlib/libs/circle/addon/lvgl/lvgl.h:38:24: note:   no known conversion for argument 1 from 'CWriteBufferDevice*' to 'CScreenDevice*'
   38 |  CLVGL (CScreenDevice *pScreen, CInterruptSystem *pInterrupt);
      |         ~~~~~~~~~~~~~~~^~~~~~~
../circle-stdlib/libs/circle/addon/lvgl/lvgl.h:35:7: note: candidate: 'constexpr CLVGL::CLVGL(const CLVGL&)'
   35 | class CLVGL
      |       ^~~~~
../circle-stdlib/libs/circle/addon/lvgl/lvgl.h:35:7: note:   candidate expects 1 argument, 2 provided
make[1]: *** [../circle-stdlib/libs/circle/Rules.mk:196: kernel.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/dolence/MiniDexed/src'
make: *** [Makefile:11: minidexed] Error 2
```

rsta2, 2024-02-09:
In MiniDexed the instance of `CScreenDevice` has the name [`mScreenUnbuffered`](https://github.com/probonopd/MiniDexed/blob/891a57809c098d77374558baa49141144771eb14/src/circle_stdlib_app.h#L135). Thus it must be `m_GUI(&mScreenUnbuffered, &mInterrupt)` in the initializer list.

dolence, 2024-02-10:
This worked! I didn't realize that there was a circle_stdlib_app.h file already initializing screen. I'm goinf to test on hardware, thank you!

smuehlst, 2024-02-25:
Looks like this was resolved, therefore I close this issue.

closed 2024-02-25 by smuehlst