smuehlst/circle-stdlib #35: No output from printf() in samples/02-stdio-hello with v16.3

swausd, 2024-05-14:
With the current release v16.3 I get no output on hdmi screen from printf() in kernel.cpp, line 36, when running samples/02-stdio-hello. Output from fprintf() from line 37 is ok.
 I get output as expected when using v16.2.

Using Raspi Zero 2 W, gcc 12.2.Rel1, 32 bit.
Am I missing some new configurations with circle step 46?

smuehlst, 2024-05-15:
Hmm, that is interesting, but I won't be able to look into this before the weekend.

smuehlst, 2024-05-18:
Reproduced the problem as described with a Raspberry Pi Zero:

* With circle-stdlib [v16.2](https://github.com/smuehlst/circle-stdlib/tree/v16.2)  the sample `02-stdio-hello` produces `stdio` output with `printf()` as expected.
* With circle-stdlib [v16.3](https://github.com/smuehlst/circle-stdlib/tree/v16.3) `stdio` output is missing, while `stderr` output still does appear on the screen.

smuehlst, 2024-05-18:
Also reproducible with a 64-bit build under QEMU.

smuehlst, 2024-05-18:
The problem is that `stdout` is for some reason no longer line-buffered. Adding a `fflush(stdout)` after `printf(...)´ makes the text appear on the screen.

closed 2024-05-18 by smuehlst

smuehlst, 2024-05-18:
@swausd I committed a fix directly to the master branch and I will create a bugfix release later today. If you want to test the fix, please make sure that you also update the circle-newlib submodule where the fix was actually implemented.

swausd, 2024-05-19:
Looks OK for me.
Thank you for the super quick fix!

commit 3bb95a59 (2024-05-18): stdout was no longer line-buffered

Fixed the problem that stdout was no longer line-buffered.

Fixes #35.