smuehlst/circle-stdlib #36: Issues with using ./configure

instrangeland, 2024-05-20:
I am using the Pi5, so I am using the 64 bit version of the arm toolchain. I have installed it. 

I have not gotten the ./configure file to run successfully. I have been using the command "./configure -r 5 -p aarch64-none-elf-" 

It gets to the point where it checks for a c compiler 

> checking for gcc... no
checking for cc... no
checking for cl.exe... no
configure: error: in `/home/eh574/cod/circle-stdlib/build/circle-newlib':
configure: error: no acceptable C compiler found in $PATH
See `config.log' for more details.

I have double checked that it is added to my path, and when I run
`which aarch64-none-elf-gcc`
It finds the right compiler. 

I double checked the output variables in the config.log file generated, and it has the correct names of the files. 

Any thoughts?

instrangeland, 2024-05-20:
Figured out that I needed to add an "export CC" to the main configure file.

smuehlst, 2024-05-20:
> I have double checked that it is added to my path, and when I run
which aarch64-none-elf-gcc
It finds the right compiler.

I cannot reproduce this. I can build successfully after adding the `aarch64-none-elf` toolchain `bin` directory to my `PATH`:

```
~/src/circle-stdlib$ which aarch64-none-elf-gcc
/home/stm/local/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/aarch64-none-elf-gcc
~/src/circle-stdlib$ ./configure -r 5 -p aarch64-none-elf-
RASPBERRYPI=5
TOOLPREFIX=aarch64-none-elf-
CIRCLE_BUILD=1
CIRCLE_QEMU=0
FLOAT_ABI=hard
NEWLIB_BUILD=1
RASPBERRYPI=5
STDLIB_SUPPORT=3
OPTIONS=
NEWLIB_INSTALL_DIR=/home/stm/src/circle-stdlib/install
NEWLIB_BUILD_DIR=/home/stm/src/circle-stdlib/build/circle-newlib
DEBUG=0
Remaining arguments:
64-bit compiler detected
STDDEF_INCPATH=/home/stm/local/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/12.2.1/include
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... aarch64-none-circle
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /usr/bin/sed
checking for gawk... no
checking for mawk... mawk
checking for gcc... gcc
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables...
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether g++ accepts -static-libstdc++ -static-libgcc... yes
checking for gnatbind... no
checking for gnatmake... no
checking whether compiler driver understands Ada... no
checking how to compare bootstrapped objects... cmp --ignore-initial=16 $$f1 $$f2
checking for objdir... .libs
checking for isl 0.16, 0.15, or deprecated 0.14... no
recommended isl version is 0.16 or 0.15, the minimum required isl version 0.14 is deprecated
checking for isl 0.16 or 0.15... no
checking for default BUILD_CONFIG...
checking for --enable-vtable-verify... no
checking for bison... no
checking for byacc... no
checking for yacc... no
checking for bison... no
checking for gm4... no
checking for gnum4... no
checking for m4... no
checking for flex... no
checking for lex... no
checking for flex... no
checking for makeinfo... no
/home/stm/src/circle-stdlib/libs/circle-newlib/missing: 81: makeinfo: not found
checking for expect... no
[... 50 lines cut ...]
checking where to find the target objcopy... pre-installed
checking where to find the target objdump... pre-installed
checking where to find the target ranlib... pre-installed
checking where to find the target readelf... pre-installed
checking where to find the target strip... pre-installed
checking where to find the target windres... pre-installed
checking where to find the target windmc... pre-installed
checking whether to enable maintainer-specific portions of Makefiles... no
configure: creating ./config.status
config.status: creating Makefile
~/src/circle-stdlib$ make
cd libs/circle && ./makeall --nosample
make[1]: Verzeichnis „/home/stm/src/circle-stdlib/libs/circle/tools“ wird betreten
  TOOL  converttool
make[1]: Verzeichnis „/home/stm/src/circle-stdlib/libs/circle/tools“ wird verlassen
make[1]: Verzeichnis „/home/stm/src/circle-stdlib/libs/circle/lib“ wird betreten
  CPP   actled.o
  CPP   alloc.o
...
<build completes successfully>
```

> Figured out that I needed to add an "export CC" to the main configure file.

I don't know why that should be necessary.

smuehlst, 2024-05-20:
> /home/eh574/cod/DTRA-URA/soc/tools/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/12.2.1/../../../../aarch64-none-elf/bin/ld: /home/eh574/cod/DTRA-URA/soc/tools/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/12.2.1/../../../../aarch64-none-elf/lib/libc.a(libc_a-exit.o): in function exit': /data/jenkins/workspace/GNU-toolchain/arm-12/src/newlib-cygwin/newlib/libc/stdlib/exit.c:65: undefined reference to _exit'

These paths look weird. In your first note it looks like you are building under the path `/home/eh574/cod/circle-stdlib`. Where do pathnames like `/data/jenkins/workspace/GNU-toolchain/arm-12/src/newlib-cygwin/newlib/libc/stdlib/exit.c` come from?

instrangeland, 2024-05-21:
I am not sure where they are coming from. These paths do not exist on my system.

smuehlst, 2024-05-21:
> I am not sure where they are coming from. These paths do not exist on my system.

Please ignore that comment where I asked about the pathnames (I deleted the comment in the meantime because I realized it was bogus). I later understood that these pathnames must have come from the system where the gcc toolchain itself was built.

As I said I cannot reproduce the problem with the documented procedure. If I shall look further into this, please provide the exact sequence of commands that reproduce the problem, starting from scratch with the recursive clone of the `circle-stdlib` repository.

instrangeland, 2024-05-21:
Thank you for the help. I created a new directory with the toolchain and a copy of the repo. 

the current path is ~/cod/test
```
git clone --recursive https://github.com/smuehlst/circle-stdlib.git
~/cod/test$ dir
arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf  circle-stdlib
~/cod/test$ cd arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin
~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin$ PATH=$PATH:~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin
~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin$ which aarch64-none-elf-gcc
/home/eh574/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/aarch64-none-elf-gcc
~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin$ cd ../
~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf$ cd ../
~/cod/test$ cd circle-stdlib/
~/cod/test/circle-stdlib$ ./configure -p aarch64-none-elf- -r 5
RASPBERRYPI=5
TOOLPREFIX=aarch64-none-elf-
CIRCLE_BUILD=1
CIRCLE_QEMU=0
FLOAT_ABI=hard
NEWLIB_BUILD=1
RASPBERRYPI=5
STDLIB_SUPPORT=3
OPTIONS=
NEWLIB_INSTALL_DIR=/home/eh574/cod/test/circle-stdlib/install
NEWLIB_BUILD_DIR=/home/eh574/cod/test/circle-stdlib/build/circle-newlib
DEBUG=0
Remaining arguments:
64-bit compiler detected
STDDEF_INCPATH=/home/eh574/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/12.2.1/include
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... aarch64-none-circle
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /usr/bin/sed
checking for gawk... no
checking for mawk... mawk
checking for gcc... no
checking for cc... no
checking for cl.exe... no
configure: error: in `/home/eh574/cod/test/circle-stdlib/build/circle-newlib':
configure: error: no acceptable C compiler found in $PATH
See `config.log' for more details.
~/cod/test/circle-stdlib$ which aarch64-none-elf-gcc
/home/eh574/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/aarch64-none-elf-gcc

```
Earlier when I added export CC to the configure file that you call, it did find the C compiler.

smuehlst, 2024-05-21:
I don't see that you export the `PATH` variable. Please repeat the steps with the difference that you execute:

```
export PATH=$PATH:~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin
```

instrangeland, 2024-05-21:
I apologize, I had run that command earlier in the terminal. But here is me running it again. 

```
~/cod/test/circle-stdlib$ export PATH=$PATH:~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin
~/cod/test/circle-stdlib$ ./configure -p aarch64-none-elf- -r 5
RASPBERRYPI=5
TOOLPREFIX=aarch64-none-elf-
CIRCLE_BUILD=1
CIRCLE_QEMU=0
FLOAT_ABI=hard
NEWLIB_BUILD=1
RASPBERRYPI=5
STDLIB_SUPPORT=3
OPTIONS=
NEWLIB_INSTALL_DIR=/home/eh574/cod/test/circle-stdlib/install
NEWLIB_BUILD_DIR=/home/eh574/cod/test/circle-stdlib/build/circle-newlib
DEBUG=0
Remaining arguments:
64-bit compiler detected
STDDEF_INCPATH=/home/eh574/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/12.2.1/include
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... aarch64-none-circle
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /usr/bin/sed
checking for gawk... no
checking for mawk... mawk
checking for gcc... no
checking for cc... no
checking for cl.exe... no
configure: error: in `/home/eh574/cod/test/circle-stdlib/build/circle-newlib':
configure: error: no acceptable C compiler found in $PATH
See `config.log' for more details.
```

```
~/cod/test/circle-stdlib$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/eh574/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin:/home/eh574/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin
```

You can see it on the end twice.

instrangeland, 2024-05-21:
I was able to get configure running, but I'm not sure why I needed these changes on my machine but not on yours.  First, on lines 208 and 209,

```
export CC
export LDFLAGS="-ffreestanding --specs=nosys.specs"

```

After that, the main configure file complained about cross compilation. For some reason, it wanted `--host "${NEWLIB_ARCH}"` in addition to the preexisting `--target "${NEWLIB_ARCH}"` on line 214. Again, not sure why this was being so odd on my machine. But after that, it ran all the way through. For now, I'll see if everything works with those changes.

smuehlst, 2024-05-21:
Ok, I see.

I compared the output of your configure run with mine, and it differs after the check for gcc.

Your run:

```
...
checking for gawk... no
checking for mawk... mawk
checking for gcc... no
checking for cc... no
checking for cl.exe... no
```

My run looks like this:

```
...
checking for gawk... no
checking for mawk... mawk
checking for gcc... gcc
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
```

It looks to me like your toolchain is not properly installed in the directory `~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf`. Are the permissions correct, e.g. is the execute bit set on the compiler executables? What happens when you simply try to invoke the compiler after setting up the `PATH` variable? Please try this:

```
~/src/circle-stdlib$ which aarch64-none-elf-gcc
/home/stm/local/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin/aarch64-none-elf-gcc
~/src/circle-stdlib$ aarch64-none-elf-gcc
aarch64-none-elf-gcc: fatal error: no input files
compilation terminated.
```

What happens in your environment?

instrangeland, 2024-05-21:
```
~/cod/test/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin$ ls -l
total 236696
-rwxr-xr-x 1 eh574 eh574   1249320 Dec  9  2022 aarch64-none-elf-addr2line
-rwxr-xr-x 1 eh574 eh574   1283232 Dec  9  2022 aarch64-none-elf-ar
-rwxr-xr-x 1 eh574 eh574   2283104 Dec  9  2022 aarch64-none-elf-as
-rwxr-xr-x 1 eh574 eh574   1270376 Dec  9  2022 aarch64-none-elf-c++
-rwxr-xr-x 1 eh574 eh574   1248168 Dec  9  2022 aarch64-none-elf-c++filt
-rwxr-xr-x 1 eh574 eh574   1266232 Dec  9  2022 aarch64-none-elf-cpp
-rwxr-xr-x 1 eh574 eh574     42960 Dec  9  2022 aarch64-none-elf-elfedit
-rwxr-xr-x 1 eh574 eh574   1270376 Dec  9  2022 aarch64-none-elf-g++
-rwxr-xr-x 1 eh574 eh574   1266192 Dec  9  2022 aarch64-none-elf-gcc
-rwxr-xr-x 1 eh574 eh574   1266192 Dec  9  2022 aarch64-none-elf-gcc-12.2.1
-rwxr-xr-x 1 eh574 eh574     35008 Dec  9  2022 aarch64-none-elf-gcc-ar
-rwxr-xr-x 1 eh574 eh574     35008 Dec  9  2022 aarch64-none-elf-gcc-nm
-rwxr-xr-x 1 eh574 eh574     35008 Dec  9  2022 aarch64-none-elf-gcc-ranlib
-rwxr-xr-x 1 eh574 eh574    761672 Dec  9  2022 aarch64-none-elf-gcov
-rwxr-xr-x 1 eh574 eh574    577664 Dec  9  2022 aarch64-none-elf-gcov-dump
-rwxr-xr-x 1 eh574 eh574    604088 Dec  9  2022 aarch64-none-elf-gcov-tool
-rwxr-xr-x 1 eh574 eh574 171300504 Dec  9  2022 aarch64-none-elf-gdb
-rwxr-xr-x 1 eh574 eh574      4627 Dec  9  2022 aarch64-none-elf-gdb-add-index
-rwxr-xr-x 1 eh574 eh574   1270808 Dec  9  2022 aarch64-none-elf-gfortran
-rwxr-xr-x 1 eh574 eh574   1316904 Dec  9  2022 aarch64-none-elf-gprof
-rwxr-xr-x 1 eh574 eh574   3394336 Dec  9  2022 aarch64-none-elf-ld
-rwxr-xr-x 1 eh574 eh574   3394336 Dec  9  2022 aarch64-none-elf-ld.bfd
-rwxr-xr-x 1 eh574 eh574  35571288 Dec  9  2022 aarch64-none-elf-lto-dump
-rwxr-xr-x 1 eh574 eh574   1269488 Dec  9  2022 aarch64-none-elf-nm
-rwxr-xr-x 1 eh574 eh574   1401056 Dec  9  2022 aarch64-none-elf-objcopy
-rwxr-xr-x 1 eh574 eh574   2655080 Dec  9  2022 aarch64-none-elf-objdump
-rwxr-xr-x 1 eh574 eh574   1283232 Dec  9  2022 aarch64-none-elf-ranlib
-rwxr-xr-x 1 eh574 eh574   1027592 Dec  9  2022 aarch64-none-elf-readelf
-rwxr-xr-x 1 eh574 eh574   1249456 Dec  9  2022 aarch64-none-elf-size
-rwxr-xr-x 1 eh574 eh574   1253560 Dec  9  2022 aarch64-none-elf-strings
```

```
~/cod/test$ aarch64-none-elf-gcc
aarch64-none-elf-gcc: fatal error: no input files
compilation terminated.

```

smuehlst, 2024-05-21:
Hmm... Do you have the normal `gcc` compiler installed? I now think that the configure step that says `checking for gcc` is trying to check for the host gcc compiler.

instrangeland, 2024-05-21:
... that did it. Sorry for the mess and thank you for helping me debug this. I just barely got this machine from IT after my last one had the main SSD go kaput.

closed 2024-05-21 by instrangeland