smuehlst/circle-stdlib #37: Synchronouse exception when running 28-touchscreen sample from /circle project

jkarstest, 2024-10-01:
Hi.
Any try to bring up the RPiTouchscreen with circle-stdlib fails (just for me?).
Independent of the template from circle_stdlib_app.h every call to the Update() method stops the programm with :

except: Synchronous exception (PC 0x9D244, EC 0x25 ISS 0x21, FAR 0x52002F, SP 0x29BC30, LR 0x80A30, SPSR 0x60000304) )

My most simple example (without using circle_stdlib_app.h): 
I copied the 28-touchscreen sample from circle-project to this projects samples folder and adapted the Makefile like this:

#
# Makefile
#

include ../../Config.mk

CIRCLEHOME = ../../libs/circle
NEWLIBDIR = ../../install/$(NEWLIB_ARCH)
#INC_DIR = $(CIRCLEHOME)/include
OBJS	= main.o kernel.o

include $(CIRCLEHOME)/Rules.mk

CFLAGS += -I "$(NEWLIBDIR)/include" -I $(STDDEF_INCPATH) -I ../../include

LIBS := "$(NEWLIBDIR)/lib/libm.a" "$(NEWLIBDIR)/lib/libc.a" "$(NEWLIBDIR)/lib/libcirclenewlib.a" \
		$(CIRCLEHOME)/addon/SDCard/libsdcard.a \
		$(CIRCLEHOME)/lib/usb/libusb.a \
	  	$(CIRCLEHOME)/lib/input/libinput.a \
	  	$(CIRCLEHOME)/addon/fatfs/libfatfs.a \
	  	$(CIRCLEHOME)/lib/fs/libfs.a \
		$(CIRCLEHOME)/lib/sched/libsched.a \
	  	$(CIRCLEHOME)/lib/libcircle.a

-include $(DEPS)

The programm 

Any Ideas ?

smuehlst, 2024-10-01:
I'm not sure that it will work to use a standard Circle sample unmodified while linking it with the circle-stdlib libraries. There are initialization steps that are executed in the base classes from `circle_stdlib_app.h` that are necessary for the integration between Circle and circle-stdlib to work. In any case this has not been tested.

rsta2, 2024-10-02:
I can reproduce the problem. The compiler does not use `memcpy()` and instead generates code, which leads to an unaligned memory access in the RPi touchscreen driver in an un-cached memory region, which generates the exception. For some reason this does not happen, when the *sample/28-touchscreen* is built without circle-stdlib. I will try to fix this soon in Circle.

rsta2, 2024-10-02:
There is an update on the *develop* branch of Circle, which should fix this problem. Until there is a new release, you have to do the following to checkout the *develop* branch, if you want to try it:

```
cd libs/circle/
git pull
git checkout develop
```

closed 2024-10-03 by jkarstest