smuehlst/circle-stdlib #39: Using circle-stdlib with Flashy?

clickworkorange, 2024-11-06:
Many thanks for the effort you've put into adding stdlib support to Circle - I cannot begin to imagine what a complex task this must be - hats off to you! Thanks to your efforts I'm able to run some third-party C++ libraries which depend on parts of stlib on a bare-metal ARM SoC, which is pretty awesome. Having migrated my project from Circle there's one thing I'm missing though: the convenience of using [Flashy](https://github.com/toptensoftware/flashy) (v3 to be specific) to deploy the compiled code. I can't wrap my head around how I would go about adding support for it, or indeed if it's even possible? Grateful for any pointers!

clickworkorange, 2024-11-06:
It even works with `USEFLASHY = 2` in `circle-stdlib/Config.mk`:
```
$ make flash
flashy \
	/dev/ttyUSB0 \
	--flashBaud:115200 \
	--userBaud:115200	 \
	--reboot:toober \
	 \
	kernel.hex
Found device: 
    - Raspberry Pi Zero
    - Serial: 00000000-fee0d14e
    - CPU Clock: 700MHz (range: 700-1000MHz)
    - Bootloader: rpi1-aarch32 v3.0.0.4, max packet size: 4096
Sending 'kernel.hex':
........................................................................................................
Transfered 393173 bytes in 35.3 seconds.
```
It's like magic! Closing...

closed 2024-11-06 by clickworkorange

smuehlst, 2024-11-06:
> I don't understand how this can work at all; I haven't provided any of the configuration settings

Well maybe the defaults are just reasonable... :-)

circle-stdlib applications are just Circle applications with some additional libraries, and the circle-stdlib samples use the standard Circle build system, so it's not so surprising that the standard Circle targets can be used.

I'm glad to hear that it all works out for you, thank you for your friendly feedback!

rsta2, 2024-11-06:
The serial bootloader support is described in libs/circle/doc/bootloader.txt.