smuehlst/circle-stdlib #41: QEMU unhandled Tags

Valefungo, 2024-12-28:
Hello,

I'm sorry to disturb again, this is something I can't handle by myself.

I added mouse support to my program, as you know I was building everything always for Raspberry 3 and reached a point where is boots and works normally until it tries to initialize the mouse, the code was picked from circle sample 10-usbmouse.

This is what happens:

Command line:
_fungostar@FungoLinux:~/baremetal$ ./qemu/build/qemu-system-aarch64 -M raspi3b -no-shutdown -action reboot=reset -action shutdown=pause -action panic=pause -action watchdog=pause -kernel circle-stdlib/samples/22-stdio-hello/kernel8.img -global bcm2835-fb.xres=1024 -global bcm2835-fb.yres=768 -d guest_errors,unimp -serial stdio -drive file=filename,if=sd,format=raw -device usb-kbd -device usb-mouse -semihosting -s -S_
bcm2835_property: unhandled tag 0x00048020
bcm2835_property: unhandled tag 0x00040010
SD: Optional CMD5 not implemented
dwc2_glbreg_write: Core soft reset not implemented
dwc2_glbreg_write: Tx FIFO flush not implemented
dwc2_glbreg_write: Rx FIFO flush not implemented
bcm2835_property: unhandled tag 0x0004800e
**bcm2835_property: unhandled tag 0x00008010**

GDB trace:

```
97		CBcmPropertyTags Tags;
99		TagSetCursorInfo.nWidth = CURSOR_WIDTH;
100		TagSetCursorInfo.nHeight = CURSOR_HEIGHT;
101		TagSetCursorInfo.nUnused = 0;
102		TagSetCursorInfo.nPixelPointer = BUS_ADDRESS ((uintptr) CursorSymbol);
103		TagSetCursorInfo.nHotspotX = CURSOR_HOTSPOT_X;
104		TagSetCursorInfo.nHotspotY = CURSOR_HOTSPOT_Y;
105		if (!Tags.GetTag (PROPTAG_SET_CURSOR_INFO, &TagSetCursorInfo, sizeof TagSetCursorInfo, 6*4))

CBcmPropertyTags::GetTag (this=0x809c898, nTagId=32784, pTag=0x809c870, nTagSize=36, nRequestParmSize=24) at bcmpropertytags.cpp:64
64		pHeader->nValueLength &= ~VALUE_LENGTH_RESPONSE;
65		if (pHeader->nValueLength == 0)
(gdb) print *pHeader
**$5 = {nTagId = 32784, nValueBufSize = 24, nValueLength = 0}**
67			return FALSE;
71	}
```

After digging this I was about to give up until I tough to rebuild everything for Raspberry 4, the real target platform.
And it got much worse :(

I'm 99% sure to have followed all the setup instructions, including the README under circle/boot to setup the sdcard image

**RPI 4**

Command line:
_fungostar@FungoLinux:~/baremetal$ ./qemu/build/qemu-system-aarch64 -M raspi4b -no-shutdown -action reboot=reset -action shutdown=pause -action panic=pause -action watchdog=pause -kernel circle-stdlib/samples/22-stdio-hello/kernel8-rpi4.img -global bcm2835-fb.xres=1024 -global bcm2835-fb.yres=768 -d guest_errors,unimp -serial stdio -drive file=filename,if=sd,format=raw -device usb-kbd -device usb-mouse -semihosting -S -s_
gic_dist_writeb: Bad offset 19c
gic_dist_writeb: Bad offset 19d
gic_dist_writeb: Bad offset 19e
gic_dist_writeb: Bad offset 19f
gic_dist_writeb: Bad offset 29c
gic_dist_writeb: Bad offset 29d
gic_dist_writeb: Bad offset 29e
gic_dist_writeb: Bad offset 29f
gic_dist_writeb: Bad offset 39c
gic_dist_writeb: Bad offset 39d
gic_dist_writeb: Bad offset 39e
gic_dist_writeb: Bad offset 39f
gic_dist_writeb: Bad offset 4e0
gic_dist_writeb: Bad offset 4e1
gic_dist_writeb: Bad offset 4e2
gic_dist_writeb: Bad offset 4e3
gic_dist_writeb: Bad offset 8e0
gic_dist_writeb: Bad offset 8e1
gic_dist_writeb: Bad offset 8e2
gic_dist_writeb: Bad offset 8e3
gic_dist_writeb: Bad offset 4e4
gic_dist_writeb: Bad offset 4e5
gic_dist_writeb: Bad offset 4e6
gic_dist_writeb: Bad offset 4e7
gic_dist_writeb: Bad offset 8e4
gic_dist_writeb: Bad offset 8e5
gic_dist_writeb: Bad offset 8e6
gic_dist_writeb: Bad offset 8e7
gic_dist_writeb: Bad offset 4e8
gic_dist_writeb: Bad offset 4e9
gic_dist_writeb: Bad offset 4ea
gic_dist_writeb: Bad offset 4eb
gic_dist_writeb: Bad offset 8e8
gic_dist_writeb: Bad offset 8e9
gic_dist_writeb: Bad offset 8ea
gic_dist_writeb: Bad offset 8eb
gic_dist_writeb: Bad offset 4ec
gic_dist_writeb: Bad offset 4ed
gic_dist_writeb: Bad offset 4ee
gic_dist_writeb: Bad offset 4ef
gic_dist_writeb: Bad offset 8ec
gic_dist_writeb: Bad offset 8ed
gic_dist_writeb: Bad offset 8ee
gic_dist_writeb: Bad offset 8ef
gic_dist_writeb: Bad offset 4f0
gic_dist_writeb: Bad offset 4f1
gic_dist_writeb: Bad offset 4f2
gic_dist_writeb: Bad offset 4f3
gic_dist_writeb: Bad offset 8f0
gic_dist_writeb: Bad offset 8f1
gic_dist_writeb: Bad offset 8f2
gic_dist_writeb: Bad offset 8f3
gic_dist_writeb: Bad offset 4f4
gic_dist_writeb: Bad offset 4f5
gic_dist_writeb: Bad offset 4f6
gic_dist_writeb: Bad offset 4f7
gic_dist_writeb: Bad offset 8f4
gic_dist_writeb: Bad offset 8f5
gic_dist_writeb: Bad offset 8f6
gic_dist_writeb: Bad offset 8f7
gic_dist_writeb: Bad offset 4f8
gic_dist_writeb: Bad offset 4f9
gic_dist_writeb: Bad offset 4fa
gic_dist_writeb: Bad offset 4fb
gic_dist_writeb: Bad offset 8f8
gic_dist_writeb: Bad offset 8f9
gic_dist_writeb: Bad offset 8fa
gic_dist_writeb: Bad offset 8fb
gic_dist_writeb: Bad offset 4fc
gic_dist_writeb: Bad offset 4fd
gic_dist_writeb: Bad offset 4fe
gic_dist_writeb: Bad offset 4ff
gic_dist_writeb: Bad offset 8fc
gic_dist_writeb: Bad offset 8fd
gic_dist_writeb: Bad offset 8fe
gic_dist_writeb: Bad offset 8ff
gic_dist_writeb: Bad offset c38
gic_dist_writeb: Bad offset c39
gic_dist_writeb: Bad offset c3a
gic_dist_writeb: Bad offset c3b
gic_dist_writeb: Bad offset c3c
gic_dist_writeb: Bad offset c3d
gic_dist_writeb: Bad offset c3e
gic_dist_writeb: Bad offset c3f
bcm2835_property: unhandled tag 0x00048013
bcm2835_property: unhandled tag 0x00048013
bcm2835_property: unhandled tag 0x00038041
bcm2835_property: unhandled tag 0x00048013

```
GDB trace:

Thread 1 hit Breakpoint 4, CStdlibAppStdio::Initialize (this=0x809b400) at ../../include/circle_stdlib_app.h:174
174	                if (!mEMMC.Initialize ())

CEMMCDevice::Initialize (this=0x809e300) at emmc.cpp:544
544		GPIOState.nGPIO = EXP_GPIO_BASE + 4;
545		GPIOState.nState = 0;
546		if (!Tags.GetTag (PROPTAG_SET_SET_GPIO_STATE, &GPIOState, sizeof GPIOState, 8))
CBcmPropertyTags::GetTag (this=0x809b348, nTagId=229441, pTag=0x809b330, nTagSize=20, nRequestParmSize=8) at bcmpropertytags.cpp:51
51		assert (pTag != 0);
52		assert (nTagSize >= sizeof (TPropertyTagSimple));
54		TPropertyTag *pHeader = (TPropertyTag *) pTag;
55		pHeader->nTagId = nTagId;
56		pHeader->nValueBufSize = nTagSize - sizeof (TPropertyTag);
57		pHeader->nValueLength = nRequestParmSize & ~VALUE_LENGTH_RESPONSE;
59		if (!GetTags (pTag, nTagSize))
(gdb) print *pHeader 
$1 = {nTagId = 229441, nValueBufSize = 8, nValueLength = 8}
64		pHeader->nValueLength &= ~VALUE_LENGTH_RESPONSE;
65		if (pHeader->nValueLength == 0)
67			return FALSE;
71	}

```

These tags are driving me crazy, I'm tempted to test this on my real rpi4 but don't really want to fry it :)

Cheers,
Val

rsta2, 2024-12-28:
QEMU does not support a mouse cursor, which is provided by the normal 
Raspberry Pi firmware. That's why the related mailbox property tags fail. You 
cannot use sample/10-usbmouse in QEMU.

The RPi 4 supports additional property tags for the support of multiple 
displays. Obviously they are also not supported by QEMU. 

I never tested Circle with QEMU with the RPi 4 emulation, which is relatively 
new.

Valefungo, 2024-12-28:
There is a way to have Circle don't try to initialize the hardware cursor ?

I need mouse support but not the actual cursor, so I can test this on the emulated rp3

testing on the real pi4 can get tedious if I have to swap sdcards at every build

rsta2, 2024-12-29:
> There is a way to have Circle don't try to initialize the hardware cursor ?

Not in Circle up to Step48. On the develop branch there is an improved display 
and also mouse support, which allows to suppress the mouse cursor with an 
additional parameter to CMouseDevice::Setup(). Unfortunately this is not 
documented yet in the Circle docs. There are only the header files and sample/
10-usbmouse for reference.

> testing on the real pi4 can get tedious if I have to swap sdcards at every
> build

Yes, but you could use the serial bootloader. You would need an USB-to-UART 
adapter for this purpose. See libs/circle/doc/bootloader.txt for more info!

Valefungo, 2024-12-29:
I'm still trying to boot qemu,
turned out it can't even reach the mouse, the very first problem is eMMC class, it can't find the inserted card and logs "no card inserted"

then just a little try with
```
#undef RASPPI
#define RASPPI 3
```
at the beginning of emmc.cpp into addon/SDCard do make it work ok and find a 3.x controller.

but after this it tries to initialize the xhcidevice and dies on the m_PCIeHostBridge (pInterruptSystem), in particular here:
```
        // PCIe init
        if (!m_PCIeHostBridge.Initialize ())
        {
                CLogger::Get ()->Write (From, LogError, "Cannot initialize PCIe host bridge");
                return FALSE;
        }
```
by reading the qemu sources, seems the PCI is not currently implemented and the RASPPI 3 trick can't work here:
```
in qemu/hw/arm/raspi4b.c
    /* Temporarily disable following devices until they are implemented */
    const char *nodes_to_remove[] = {
        "brcm,bcm2711-pcie",
        "brcm,bcm2711-rng200",
        "brcm,bcm2711-thermal",
        "brcm,bcm2711-genet-v5",
    };

```

so I'm back to the beginning.
Will try to find an USB to UART adapter, swapping sdcards is way too tedious after the first 2 times
marking closed this time :)

@rsta2 @smuehlst would be of any interest if, after making it working, I push this thing as a new "sample" for circle-stdlib ?
I think you've already got that its the adaptation of an x86 emulator

closed 2024-12-29 by Valefungo

rsta2, 2024-12-29:
SD card access is different on the RPi 4. And if PCIe is not emulated, the xHCI USB host won't work and thus also the USB mouse.

smuehlst, 2024-12-30:
> would be of any interest if, after making it working, I push this thing as a new "sample" for circle-stdlib ?
> I think you've already got that its the adaptation of an x86 emulator

@Valefungo I'm not sure what exactly you mean with that. If you are suggesting that this is added added as a new sample to the circle-stdib repository, I don't think a whole x86 emulator would be reasonable. So far I tried to keep the samples small to demonstrate and test the basic features of the C and C++ standard libraries. But thanks for the offer!

Of course you are free to provide a separate repository where you integrate Circle and circle-stdlib into the build of your x86 emulator.