smuehlst/circle-stdlib #43: It seems there is something wrong when readline from stdin when typed "shift+something"

zsdz, 2025-05-13:
I try to run the sample 02,after typing (+ 1 1) from keyboard,the sample echo 

) ` 1 1=

....

It seems there is something wrong when readline from stdin when using the key shift?

Thanks!

rsta2, 2025-05-13:
On which Raspberry Pi does this happen?

I did a quick test in QEMU with RPi 1 and RPi 3 (64-bit) emulation and it worked:

[image]

zsdz, 2025-05-14:
I use rpi4b,real machine.and I think there is the problem,because I change the keyboard,but got the same thing

rsta2, 2025-05-14:
I tested it here on a real RPi 4B (AArch64) and it worked too. Which revision 
code is display in the 2nd line on screen? Does this happen also with the 
small program, I have posted here:

https://github.com/rsta2/circle/discussions/574#discussioncomment-13086911

zsdz, 2025-05-15:
1 after poweron rpi4b with circle-stdlib sample2,the screen move fast,so I take a vedio,the first 2 line is:

logger:Circle 49 started on Raspberry Pi 4 Model B 1GB(AArch64)
logger:Revision code is a03115,compiler has been GCC 13.2.1

(this kernel8.img is build in ubuntu)

2 your console-simple I got the same error too...shift+numkey is wrong...so the problem is my rpi4 or keyboard?my new 4g rpi4b is on the road,I can test again

3 I check the configure file again:

circle Rules.mk(I am not change configure file,direct create Rules.mk):
AARCH = 64
RASPPI = 4
PREFIX64 = aarch64-none-elf-

circle-stdlib configure(the readme.md didn't tell me to change it,but the defalut RASPBERRYPI is 1):
RASPBERRYPI=4
CIRCLE_BUILD=49

so it seems no error here...

4 I doubt is it because I build it under WSL?so I build it on ubuntu again,but got the same error...

so there must be something wrong...but I don't know why...

if I want to type (,I have to press key "shift+8",not "shift+9"

5 my sd car is 32G format to fat32,has these files:config.txt,cmdline.txt,bcm2711-rpi-4-dtb,fixup4,
start4.elf,kernel8.img

[kernel8CircleStdlibSample02.txt](https://github.com/user-attachments/files/20220218/kernel8CircleStdlibSample02.txt)(chang .txt to .img)

rsta2, 2025-05-15:
Thanks for the comprehensive info! Now it's getting clearer. I think this is simply a keymap problem, and it is caused by the issue raspberrypi/firmware#1968. My guess is, that the response size of the result for the property mailbox call, when fetching the command line, is wrong, and thus the `keymap=US` option is not processed.

I guess it's enough, when you download the most recent firmware. You can do this by checking out the *develop* branch of Circle, go to *boot/*, modify the *Makefile* so that the `FIRMWARE=` variable is set to `master` and doing `make` there. Install the downloaded files on you SD card.

For a quick test with the old firmware, you can also add the option 
`-o DEFAULT_KEYMAP=\"US\"`, when you call `./configure` for the circle-stdlib project.

zsdz, 2025-05-16:
Thanks!I did as you say,but there seems some network connect error when fetching the firmware file using make command.so I direct download the file from 

https://github.com/raspberrypi/firmware/tree/master

then copy the content of boot floder(the files date is 2025.5.14 4:30) to sd card.The firmware file should be newest.I use your console-simple prog to test,but the error is the same...

I recompile the console-simple prog,but it seems it's not necessary to recompile

I am not familiar with the firmware file of rpi4b...Do I have to copy so many file to sd card?Can't I copy few file like 

https://bgithub.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/05_drivers_gpio_uart#rpi-4

said?after all I only need to use the basic I(usbkeyboard)O(Screen output),and I can type most key right.And I use the official rpi os to test,change the img to my test img,but the error is the same

maybe I should fix the problem later...now I can use some key not direct:
shift+8 for (
] for +
...

rsta2, 2025-05-16:
I can only say, that the US keymap works here with samples/02-stdio-hello on 
an early 2GB RPi 4B (revision code b03111) with the recommended firmware 
(downloadable in libs/circle/boot/ after checking out the develop branch 
there) and the bootloader in EEPROM from May 8 2025, when there is a 
cmdline.txt file on the SD card with this contents:

keymap=US

Further files needed for the RPi 4 in AArch64 mode on the SD card are:

armstub8-rpi4.bin
bcm2711-rpi-4-b.dtb
config.txt
fixup4.dat
kernel8-rpi4.img
start4.elf

zsdz, 2025-05-19:
I change circle-stdlib/libs/circle/lib/koptions.cpp line 47 (in function CKernelOptions::CKernelOptions (void))

from 

    strcpy (m_KeyMap, DEFAULT_KEYMAP);

to

    strcpy (m_KeyMap, "US");

and circle-stdlib/libs/circle/lib/input/keymap.cpp line 143 (in function CKeyMap::CKeyMap (void))

from 

    pDefaultMap = LookupDefaultMap (DEFAULT_KEYMAP);

to
  
    pDefaultMap = LookupDefaultMap ("US");

then comment line 

    assert (pDefaultMap != 0);

next to it

and fix the problem...

it may not be a good way,but it works now...

and if I change /circle/include/circle line 298,comment the default 

    #define DEFAULT_KEYMAP          "DE"

then uncomment the 

    #define DEFAULT_KEYMAP          "US"

the sample 08-usbkeyboard will work for some shift+key that can't work before,but can work fine for the key "back"...

zsdz, 2025-05-19:
I found the problem...my cmdline.txt file format is wrong,the options are connect use space key,not enter like config.txt
I am sorry....I think you can delete this issuse