smuehlst/circle-stdlib #50: Hangs On All Devices (including QEMU)

TTWNO, 2025-08-06:
Following the steps provided in the `README.md` file produces examples that halt or crash after the `CKernel::Initialize` step. I have attempted the following matrix:

- Toolchains:
    - gcc 13.2.Rel1 (aarch64-none-elf-)
    - gcc 14.3.Rel1 (aarch64-none-elf-)
    - gcc 13.2.Rel1 (arm-none-eabi-)
    - gcc 14.3.Rel1 (arm-none-eabi-)
- Versions of Circle stdlib:
    - `master`
    - `develop`

None of these produce binaries that function correctly either under QEMU or my physical RasPi 4b. It seems that it runs the `CKernel::Initialization` code, and then halt—this indicates that the initialization fails (fair enough) but not enough info is given by QEMU (and the physical device is even harder) in order to debug what is actually failing.

When attempting with 4B in QEMU, the log halts and does nothing after the `Raspberry Pi 4, Circle` etc. messages. When attempting with 3B in QEMU, QEMU itself exists immediately after initialization, but I can briefly see some type of log output before it stops.... I'm assuming this is the same output.

I have tried following the steps form CI in order to reproduce the results you get in Actions. All attempts result in the same trial and error, immediate closing of QEMU, or a halt after basic init meessages are logged.

The physical Raspberry Pi 4B displays the same behaviour. When adding LED flashes both before, during, and just before return of `CKernel::Run` shows that only initialization is ever run, then the program halts. I've used Circle's `make install64/32`, I've tried manually moving only the required files myself. All result in exactly the same behaviour.

```bash
# download arm toolchain
export PATH="$PATH:arm-toolchain/bin/"
git clone --recursive https://github.com/smuehlst/circle-stdlib
cd circle-stdlib
./configure -r 4 -p arm-toolchain-
make
make build-samples
# mount SD card at /mnt
# move basic files/all files via `make install64/32` to /mnt
cp sample/04-std/kernel*img /mnt
# unmount /mnt 
```

I feel like I'm probably missing a fundamental step here that is generally know and just not documented. But if I'm doing everything right, then this might be a bug.

TTWNO, 2025-08-07:
> It is important that you use the modified version of QEMU from this repository: https://github.com/smuehlst/qemu

Ah! That is what I was missing! I got the Smoke screen examples working in QEMU. It still doesn't run on the physical 4B, but interestingly, doesn't run with the `-M raspi4b` flag in QEMU either. I tried both GCC 14 and 13. So maybe there is an issue with the Pi4 integration. Pi 3 64-bit and Pi 0 32-bit work in QEMU, but I don't have a corresponding physical device to verify running on real hardware.

> Is arm-toolchain- used literally? Or did you mean this as a placeholder for the actual toolchain prefix?

Just a placeholder :)

rsta2, 2025-08-20:
You can do one of two things:

1. You can specify the option "-o SCREEN_HEADLESS", when calling "configure". 
This will put the class CScreenDevice into a dummy mode, so that it excepts 
calls to Initialize() and Write(), but simply ignores them.

2. You can remove the "mScreen" member variable from the class 
CStdlibAppScreen in circle-stdlib/include/circle_stdlib_app.h and include the 
modified file into your application.